open62541/open62541-nodeset-loader #24: NodeSet with multiple namespaces

MartinaFelberFEM, 2020-04-23:
Support for multiple namespaces needed?
1 NodeSet contains 2 or more new namespaces.
E.g.: Euromap83 and Euromap77 within 1 XML

matkonnerth, 2020-04-24:
I think this requirement should be supported by the loader. At the moment, the abstraction called "Nodeset" is more or less a nodeset.xml which could contain 1-n namespaces. All nodes in the file are parsed, sorted and then for example added to the open server.

I've made a test case for this requirement: https://github.com/matkonnerth/nodesetLoader/pull/30
Please take a look, if this is what you meant.

The interesting thing is:
When both namespaces are in 1 file, sorting will occur across both namespaces and even the case that instance nodes are in the file before types is handled.

If the type namespace and instance namespace is split across 2 files, the nodesetLoader will rely on the user that the types are load before the instances. For me at the moment I see no use case to "merge" multiple files into one Nodeset. What do you think?

MartinaFelberFEM, 2020-04-26:
Looks very good. Thanks
Great that sorting across both namespaces works too.
I think it's not necessary to merge multiple files into one nodeset. The use case you have tested is the one I meant.

closed 2020-04-26 by MartinaFelberFEM