open62541/open62541-nodeset-loader #57: PLCopen nodeset import does not work

MartinaFelberFEM, 2020-05-06:
It seems that PLCopen nodeset import does not work anymore ? 
I have tested with current master. f238d09a6f2cde1baeb5e397395e9339d27c9860

Function NodesetLoader_loadFile() returns true
There are no error messages or warnings.

Nodeset DI and testNodeSet.xml could be loaded.
But PLCopen namespace is missing

[image]

Can you confirm?

matkonnerth, 2020-05-06:
ok, that looks interesting. I've tested it with _backends/open62541/examples/server /usr/local/share/open62541/tools/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml /usr/local/share/open62541/tools/ua-nodeset/PLCopen/Opc.Ua.Plc.NodeSet2.xml_

here's the result:
[image]

MartinaFelberFEM, 2020-05-06:
I am very sorry, I have used the wrong nodeset (OpcUaPLCopenModel.xml) 
:(
Sorry for the inconvenience.

closed 2020-05-06 by MartinaFelberFEM