open62541/open62541-nodeset-loader #59: ReferenceTypes: Attributes InverseName and Symmetric

MartinaFelberFEM, 2020-05-07:
Hi!

I think the referencetype attributes InverseName and Symmetric are not properly set.

reference server:
[image]
Symmetric = false

test server:
[image]
Symmetric = true

Other examples:
DI nodeset:
Id = ns=2;i=6031 REFERENCETYPE
BrowseName = ns=2;IsOnline

PLCopen:
Id = ns=3;i=4001 REFERENCETYPE
Id = ns=3;i=4002 REFERENCETYPE
Id = ns=3;i=4003 REFERENCETYPE
Id = ns=3;i=4004 REFERENCETYPE
...

matkonnerth, 2020-05-08:
Yes, you are right. Here the fix: https://github.com/matkonnerth/nodesetLoader/pull/60

MartinaFelberFEM, 2020-05-08:
Great! I have just tested it with current master branch. :)

Now I only have one difference left for nodeSets DI and PLCopen!

closed 2020-05-08 by MartinaFelberFEM