open62541/open62541-nodeset-loader #77: DataType nodes: DataTypeDefinition test

MartinaFelberFEM, 2020-05-10:
Summary of open questions/issues regarding DataType nodes and their DataTypeDefinition attribute:

Regarding to OpcUa spec 3 the DataTypeDefinition attribute is optional unless the node is

- derived from Structure
- derived from Enumeration
- derived from OptionSet
- derived from Unsigned integer and provides a HasProperty reference with OptionSetValues reference

[image]

It seems, that namespace 0 violates the specification, because following nodes of namespace 0, do not provide the DataTypeDefinition attribute:

- all types derived from Enumeration
- the DataTypeDefinition structure itself -> we are not sure if this is the case, because the DataType is abstract
[image]

- in general we are not sure if abstract DataType nodes have to provide this attribute or not

MartinaFelberFEM, 2020-05-10:
DI namespace: nodeset compiler vs. nodeset loader

The nodeset compiler does not provide a DataTypeDefinition attribute for the defined structures.
Therefore the address space comparison does not work.

The output of the nodeset loader seems correct.

Structures with arrays:
We are not sure if the LengthField shall be visible as structure field in addressspace, but probably not. 

e.g.
Opc.Ua.Di.Types.bsd defines the LengthField - NoOfParameterDefs

```xml
  <opc:StructuredType Name="TransferResultDataDataType" BaseType="tns:FetchResultDataType">
    <opc:Field Name="SequenceNumber" TypeName="opc:Int32" />
    <opc:Field Name="EndOfResults" TypeName="opc:Boolean" />
    <opc:Field Name="NoOfParameterDefs" TypeName="opc:Int32" />
    <opc:Field Name="ParameterDefs" TypeName="tns:ParameterResultDataType" LengthField="NoOfParameterDefs" />
  </opc:StructuredType>
```

vs.
Opc.Ua.Di.Nodeset2.xml: declares ParameterDefs as array

```xml
  <UADataType NodeId="ns=1;i=15889" BrowseName="1:TransferResultDataDataType">
    <DisplayName>TransferResultDataDataType</DisplayName>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">ns=1;i=6522</Reference>
    </References>
    <Definition Name="1:TransferResultDataDataType">
      <Field Name="SequenceNumber" DataType="i=6" />
      <Field Name="EndOfResults" DataType="i=1" />
      <Field Name="ParameterDefs" DataType="ns=1;i=6525" ValueRank="1" />
    </Definition>
  </UADataType>
```

We think that it should not be visible, that it's only used for internal struct type instantiation. But we are not sure about it.

MartinaFelberFEM, 2020-05-10:
Test: Instantiate a variable in both servers of type TransferResultDataDataType

We haven't been able to instantiate a variable of type TransferResultDataDataType regardless of the server implementation:

- reference server with nodeset compiler and generated src code
- test server with nodeset loader using `getCustomDataType` function

matkonnerth, 2020-05-10:
size of arrays: is needed for encoding/decoding, but is not needed for user of the datatype, because the length field is implicitly set/calculated. But the 2 different bsds are interesting.

It should be possible to use the TransferResultDataType, for example with:

UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.dataType = UA_NODEID_NUMERIC(2, 15889);
    UA_Server_addVariableNode(
        server, UA_NODEID_NUMERIC(1, 1001), UA_NODEID_NUMERIC(0, 85),
        UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
        UA_QUALIFIEDNAME(1, "myVar"),
        UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), attr, NULL, NULL);

closed 2020-05-25 by MartinaFelberFEM