open62541/open62541-nodeset-loader #78: Euromap test: EventNotifier attribute

MartinaFelberFEM, 2020-05-11:
I think the nodesetLoader does not set the EventNotifier attribute correctly.
At least it does not match with the reference server.
Tested with branch euormap_test

E.g. Object ProductionDatasetLists: BaseObjectType/ProductionDataSetManagementType/ProductionDataSetLists

reference server: EventNotifier = SubscribeToEvents
[image]

test server: EventNotifier = None
[image]

xml Definition: Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml
```xml
    <UAObject ParentNodeId="ns=1;i=1008" EventNotifier="1" NodeId="ns=1;i=5007" BrowseName="1:ProductionDatasetLists">
        <DisplayName>ProductionDatasetLists</DisplayName>
        <Description>Functions for exchanging information on the available production datasets on client and server</Description>
        <References>
            <Reference ReferenceType="HasComponent">ns=1;i=7005</Reference>
            <Reference ReferenceType="HasModellingRule">i=80</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1008</Reference>
            <Reference ReferenceType="HasTypeDefinition">ns=1;i=1003</Reference>
            <Reference ReferenceType="GeneratesEvent">ns=1;i=1040</Reference>
            <Reference ReferenceType="HasComponent">ns=1;i=7006</Reference>
        </References>
    </UAObject>
```

The XML definition does not specify the EventNotifier attribute ...
How can you know if you have to set the EventNotifier attribute?
Maybe you have to set it, if the node is of class Object and it has a GeneratesEvent reference?

How can you know if you have to set the EventNotifier attribute for historizing events? (HistoryRead/HistoryWrite)

matkonnerth, 2020-05-12:
I can confirm, EventNotifier attribute is currently not parsed, simply have to add it, wanna try? ;)

Don't understand your questions, in the your xml snippet, there is the attribute:
EventNotifier="1"

MartinaFelberFEM, 2020-05-12:
Ah sorry, I have totally overlooked the EventNotifier attribute in the XML

Yes, I would like to try :)
I will create a PR for EventNotifiers

matkonnerth, 2020-05-15:
fixed by https://github.com/matkonnerth/nodesetLoader/pull/89

closed 2020-05-15 by matkonnerth