open62541/open62541-nodeset-loader #79: Euromap test: "double free or corruption" at server shutdown

MartinaFelberFEM, 2020-05-11:
Tested with euromap_test branch.
`
./testServer "$NODESET_PATH_OPEN62541/DI/Opc.Ua.Di.NodeSet2.xml"     "$NODESET_PATH_OPEN62541/PLCopen/Opc.Ua.Plc.NodeSet2.xml"     "$NODESET_PATH_BACKEND_TESTS/euromap/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml"     "$NODESET_PATH_BACKEND_TESTS/euromap/Opc.Ua.PlasticsRubber.IMM2MES.NodeSet2.xml"`

Get following output at server shutdown:
[2020-05-11 21:13:06.427 (UTC+0200)] info/network	Shutting down the TCP network layer
**double free or corruption (!prev)**
Aborted (core dumped)

This does not happen when I only load DI and PLCopen nodesets.

valgrind command and output:
`valgrind --tool=memcheck --leak-check=full --track-origins=yes --undef-value-errors=yes --show-leak-kinds=all --leak-resolution=high --num-callers=30 ./testServer $NODESET_PATH_OPEN62541/DI/Opc.Ua.Di.NodeSet2.xml $NODESET_PATH_OPEN62541/PLCopen/Opc.Ua.Plc.NodeSet2.xml $NODESET_PATH_BACKEND_TESTS/euromap/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml $NODESET_PATH_BACKEND_TESTS/euromap/Opc.Ua.PlasticsRubber.IMM2MES.NodeSet2.xml`

[valgrind_euromap_double_free.txt](https://github.com/matkonnerth/nodesetLoader/files/4611627/valgrind_euromap_double_free.txt)

It seems that there are invalid reads/writes

matkonnerth, 2020-05-12:
hm, I've tried to reproduce but I couldn't.
I've tested it with _backends/open62541/examples/server ../nodesets/Opc.Ua.Di.NodeSet2.xml ../nodesets/Opc.Ua.Plc.NodeSet2.xml  ../nodesets/euromap/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml ../nodesets/euromap/Opc.Ua.PlasticsRubber.IMM2MES.NodeSet2.xml_

I'm on x86-64, gcc 8.3, I've tried it in release and debug and current master f200ecc.

matkonnerth, 2020-05-12:
This output looks interesting (from your logfile), I think he is cleaning up memory of a Variant. That variant was set while adding a variable node from xml.

==28425== Invalid write of size 8
==28425==    at 0x4C36657: memset (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==28425==    by 0x52F9A4E: UA_Array_delete (ua_types.c:1287)
==28425==    by 0x52FA8AA: Variant_clear.part.8 (ua_types.c:523)
==28425==    by 0x52F906C: UA_clear (ua_types.c:1220)
==28425==    by 0x52F1A0E: deleteNodeMapEntry (ua_nodestore_hashmap.c:186)
==28425==    by 0x52F1A0E: UA_NodeMap_delete (ua_nodestore_hashmap.c:420)
==28425==    by 0x52F9221: UA_ServerConfig_clean (ua_server_config.c:58)
==28425==    by 0x52F9C85: UA_Server_delete (ua_server.c:223)
==28425==    by 0x108E6B: main (server.cpp:40)
==28425==  Address 0x9ee1c50 is 0 bytes inside a block of size 4 alloc'd
==28425==    at 0x4C31B25: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==28425==    by 0x52FAEAE: UA_Array_copy (ua_types.c:1258)
==28425==    by 0x52FB36E: Variant_copy (ua_types.c:535)
==28425==    by 0x52FAD98: UA_copy (ua_types.c:1121)
==28425==    by 0x52FAFE0: UA_Variant_copy (types_generated_handling.h:744)
==28425==    by 0x52FAFE0: copyCommonVariableAttributes (ua_nodes.c:400)
==28425==    by 0x52FC8DA: copyVariableNodeAttributes (ua_nodes.c:413)
==28425==    by 0x52FC8DA: UA_Node_setAttributes (ua_nodes.c:485)
==28425==    by 0x530CDFE: AddNode_raw (ua_services_nodemanagement.c:906)
==28425==    by 0x530E896: Operation_addNode_begin.lto_priv.2755 (ua_services_nodemanagement.c:1009)
==28425==    by 0x531629C: UA_Server_addNode_begin (ua_services_nodemanagement.c:1384)
==28425==    by 0x4E4391F: handleVariableNode (import.c:195)
==28425==    by 0x4E44377: addNode (import.c:330)
==28425==    by 0x4E44940: addNodes (import.c:433)
==28425==    by 0x4E44BDC: NodesetLoader_loadFile (import.c:486)
==28425==    by 0x108E29: main (server.cpp:32)

or here

==28425== Invalid write of size 8
==28425==    at 0x4C3665A: memset (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==28425==    by 0x52F9A4E: UA_Array_delete (ua_types.c:1287)
==28425==    by 0x52FA8AA: Variant_clear.part.8 (ua_types.c:523)
==28425==    by 0x52F906C: UA_clear (ua_types.c:1220)
==28425==    by 0x52F1A0E: deleteNodeMapEntry (ua_nodestore_hashmap.c:186)
==28425==    by 0x52F1A0E: UA_NodeMap_delete (ua_nodestore_hashmap.c:420)
==28425==    by 0x52F9221: UA_ServerConfig_clean (ua_server_config.c:58)
==28425==    by 0x52F9C85: UA_Server_delete (ua_server.c:223)
==28425==    by 0x108E6B: main (server.cpp:40)
==28425==  Address 0x9ee1c58 is 4 bytes after a block of size 4 alloc'd
==28425==    at 0x4C31B25: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==28425==    by 0x52FAEAE: UA_Array_copy (ua_types.c:1258)
==28425==    by 0x52FB36E: Variant_copy (ua_types.c:535)
==28425==    by 0x52FAD98: UA_copy (ua_types.c:1121)
==28425==    by 0x52FAFE0: UA_Variant_copy (types_generated_handling.h:744)
==28425==    by 0x52FAFE0: copyCommonVariableAttributes (ua_nodes.c:400)
==28425==    by 0x52FC8DA: copyVariableNodeAttributes (ua_nodes.c:413)
==28425==    by 0x52FC8DA: UA_Node_setAttributes (ua_nodes.c:485)
==28425==    by 0x530CDFE: AddNode_raw (ua_services_nodemanagement.c:906)
==28425==    by 0x530E896: Operation_addNode_begin.lto_priv.2755 (ua_services_nodemanagement.c:1009)
==28425==    by 0x531629C: UA_Server_addNode_begin (ua_services_nodemanagement.c:1384)
==28425==    by 0x4E4391F: handleVariableNode (import.c:195)
==28425==    by 0x4E44377: addNode (import.c:330)
==28425==    by 0x4E44940: addNodes (import.c:433)
==28425==    by 0x4E44BDC: NodesetLoader_loadFile (import.c:486)
==28425==    by 0x108E29: main (server.cpp:32)

Maybe there I'm trashing some memory.

MartinaFelberFEM, 2020-05-12:
I have used the latest euromap release on my test branch. There are some differences in the XMLs.
But I haven't had a deeper look.

matkonnerth, 2020-05-12:
oh, ok, can you make an new PR for this test?

MartinaFelberFEM, 2020-05-12:
Yes, sure.
But my test is not finished yet. And I am not sure how my client browse issues are related to this. So I wouldn't merge it into master right now.

matkonnerth, 2020-05-12:
It's no problem if it isn't ready or isn't building at the moment, but now I can take a look on it.

matkonnerth, 2020-05-12:
okay, can reproduce it with your Pr. Has definitely todo with value import, when I disable value import, everything works fine.

matkonnerth, 2020-05-13:
Problem was, that with every new nodesets that brings in new datatypes, the customTypes array was reallocated, but ptrs to this datatypes were already used by variant values. So on destruction of this values, the type ptr was dangling. Current solution is, that we allocate a fixed size of memory for the DataTypes. Maybe this can be done in a more clever way, but the open internal switches only between internal (namespace 0) and customTypes with a boolean flag.

closed 2020-05-13 by matkonnerth