open62541/open62541-nodeset-loader #80: Euromap test: test client issues

MartinaFelberFEM, 2020-05-11:
**Reference Server:**

Addressspace output of reference server seems strange, as there are some nodeIds twice in the output.
E.g.:
Id = ns=4;i=6012 VARIABLE
BrowseName = ns=4;RemoteUserName

**Test Server:**
BrowseResponse - responseHeader serviceResult is Bad (BadDecodingError) when browsing the test server address space.

Error BrowseReferences failed. Id = ns=4;i=6013
Test finished: failure

This starts one numeric NodeId after the strange double Ids in the output? ns=4;i=6012

matkonnerth, 2020-05-12:
I think the decoding error could relate to the memory issue.

matkonnerth, 2020-05-13:
also the nodes with id 6012, 6013, 6013, 6014 is twice in the nodeset, both of reference and testServer.

MartinaFelberFEM, 2020-05-14:
Thanks for your comments. Yes, I have noticed the duplicate nodeIds
Sry, for my late reply. I'll start fixing it this afternoon.

matkonnerth, 2020-05-15:
enumValues issue is on server side, here the fix: https://github.com/matkonnerth/nodesetLoader/pull/90

MartinaFelberFEM, 2020-05-17:
Found source of duplicate Ids in server addressspace output:

If we have a look at the referenceServer output we often see NodeIds twice in the output:
E.g. 

**Id = ns=3;i=1004** OBJECTTYPE
BrowseName = **ns=3;MessageConditionType**
DisplayName = :MessageConditionType
Description = :Text messages (incl. error messages) of the control system currently shown on the screen of the machine
WriteMask = 0
UserWriteMask = 0
IsAbstract = false
References: 
	| i=46 ; ns=3;i=6001 | 
	| i=46 ; ns=3;i=6009 | 
	| i=46 ; ns=3;i=6015 | 

**Id = ns=3;i=1004** OBJECTTYPE
BrowseName = **ns=3;MessageConditionType**
DisplayName = :MessageConditionType
Description = :Text messages (incl. error messages) of the control system currently shown on the screen of the machine
WriteMask = 0
UserWriteMask = 0
IsAbstract = false
References: 
	| i=46 ; ns=3;i=6001 | 
	| i=46 ; ns=3;i=6009 | 
	| i=46 ; ns=3;i=6015 | 

As the NodeIds are sorted, we loose the parent hierarchy.
We have added some traces to follow the hierarchy and check why there are duplicates in the output.

E.g. for Nodeid -  **ns=3;i=1004 MessageConditionType**

1. 
    copy id = ns=3;i=1004   // MessageConditionType
    Parent Id = i=2915

2. 
    copy id = ns=3;i=1004
    Parent Id = i=2915

Follow: Id = i=2915     // AlarmConditionType

1.
    copy id = i=2915
    Parent Id = i=2881
2.
    copy id = i=2915
    Parent Id = i=2881

Follow: Id = i=2881     // AcknowledgeableConditionType

1.
    copy id = i=2881
    Parent Id = i=2782
2.
    copy id = i=2881
    Parent Id = i=2782

Follow: Id = i=2782     // ConditionType

1.
    copy id = i=2782
    Parent Id = i=2041
2.
    copy id = i=2782
    Parent Id = i=2041

Follow: i=2041          // BaseEventType

1.
    copy id = i=2041
    Parent Id = i=58

2.
    copy id = i=2041
    Parent Id = i=3048

**Now we have 2 different parents!**

Parent Id = i=58    // BaseObjectType
Parent Id = i=3048  // EvenTypes Folder

It seems that BaseEventType has an inverse reference HasSupertype to BaseObjectType

[image]

As HasSubType (HasSuperType) reference is a hierarchical reference, we probably get 2 times the browse result. That's why we print the duplicates.

[image]

Maybe we should remove the includeSubTypes flag at the browseRequest ...

MartinaFelberFEM, 2020-05-17:
Hmm, we can't set includeSubTypes to false, because we need all hierarchical references (Organizes, HasComponent, etc.) ...

nodeClassMask:
we would like to browse all nodeClasses ...

Hmm, can't think of way to avoid the duplicates

@matkonnerth : Would it be okay if we have duplictes in the output (for all EventTypes) ?

matkonnerth, 2020-05-17:
My understanding of the algorithm is, that it starts at the root node and browses all hierachical references in forward direction to get all nodes. Then for these nodes the attributes and non-hierachical references are written to the output.

You are right, the BaseEventType has 2 hierachical references "pointing" at him. To solve this, we could use a set for the nodeContainer.
For me it's okay to have the duplicates in the output at the moment.

MartinaFelberFEM, 2020-05-17:
Thanks for your suggestion to use a set!

I hope that we don't loose any information if we use a set, or if we just add all nodeIds once ...

But it should be no problem, because the references to the same nodeId are in the output. And if one would be missing we would see it. 

Id = i=58 OBJECTTYPE
BrowseName = ns=0;BaseObjectType
DisplayName = :BaseObjectType
Description = :
WriteMask = 0
UserWriteMask = 0
IsAbstract = false
References: 
	| i=45 ; i=61 | 
	| i=45 ; i=75 | 
	| i=45 ; i=76 | 
	| i=45 ; i=77 | 
	| i=45 ; i=2004 | 
	| i=45 ; i=2013 | 
	| i=45 ; i=2020 | 
	| i=45 ; i=2026 | 
	| i=45 ; i=2029 | 
	| i=45 ; i=2033 | 
	| i=45 ; i=2034 | 
	**| i=45 ; i=2041 |** 
        ........

Id = i=3048 OBJECT
BrowseName = ns=0;EventTypes
DisplayName = :EventTypes
Description = :
WriteMask = 0
UserWriteMask = 0
EventNotifier = 0
References: 
	**| i=35 ; i=2041 |** 
	| i=40 ; i=61 | 

Should be okay

matkonnerth, 2021-06-03:
doesn't harm to have the types twice in the output.

closed 2021-06-03 by matkonnerth