open62541/open62541-nodeset-loader #94: openBackend: interface dataType import

matkonnerth, 2020-05-17:
Currently all memory that the nodesetLoader allocates is also freed by himself, but not in the case, when DataTypes are imported. In this case the, the DataTypeImporter allocates memory for the custom dataTypes and sets the ptr of serverconfig->customDataTypes. The nodesetLoader cannot free up this memory, because it's needed later on for encoding/decoding.

So the question is, if the user of the library should setup upfront the customDataTypeArray and the nodesetLoader uses this memory. If no memory is provided, no dataTypes will be imported (the nodes will be imported, but there will be no UA_DataType for it). 

This requires no change of the interface, because the memory could be provided via serverConfig->customDataTypes.

MartinaFelberFEM, 2020-05-18:
How shall the user know how much memory is necessary to allocate? 
Shall the user guess the necessary size (e.g. 100 times sizeof(UA_DataType)) as you did at DataTypeImporter_addCustomDataType() and if the NodeSetLoader returns no error everything is ok and the size was big enough? 
The user could probably calculate the size beforehand if he counts the StructuredType definitions in the .bsd files ...

I think it would be okay to make the user responsible for allocating/freeing the memory.
The open62541 SDK can't know if the memory is on stack or on heap to free it correctly.
I think at the custom data type examples of open62541 SDK the user too has either to allocate the memory on heap or provide an array on stack.

Maybe it would be easier for the user if the serverConfig provides one pointer for customDataTypes and one for the NodeSetLoader? Just in case the user want's both. But then you would have to change the interface and it's not really necessary ...

What I don't understand: How does the server find the DataTypes of the nodeset-compiler generated types? E.g. UA_TYPES_INTEGRATION_TEST_DI array.
Is it appended to the UA_TYPES Array?
Can custom data types, nodesetLoader types and nodeset-compiler types all coexist?

matkonnerth, 2020-05-18:
These are good questions. You are right, the user doesn't know how many types will be imported. So for a preallocated array we would also need the maximum size, that we must not exceed.

Regarding the last questions:
It's not appended to the types array. If you are setting up a variant, the variant has a ptr to the UA_Datatype used, so the server finds the type. For reading the datatypedefinition the generated UA_DataType array has to be appended to the serverconfig->customTypes. Yes, I think the can coexist. What I know is, that all custom datatypes should be in 1 array (https://github.com/open62541/open62541/issues/2808), so if there are existing custom types (maybe generated by the nodeset compiler), we would have to append it to existing custom datatype array.

matkonnerth, 2020-06-26:
interface will stay as it is, documentation was added that the user has to manually clean up the custom datatypes

closed 2020-06-26 by matkonnerth