open62541/open62541-nodeset-loader #97: EM83: Integration test found difference in JobListElementType

matkonnerth, 2020-05-18:
Now it gets interesting, the integration test looks with a big magnifier onto the implementations;)

reference server:

[ 
		Name = PlannedProductionTime
		Description = :
		DataType = i=11
		ValueRank = -1
		ArrayDimensions = [] 
		MaxStringLength = 0
		IsOptional = 0
	] 

[ 
		Name = PlannedProductionTime
		Description = :
		DataType = i=290
		ValueRank = -1
		ArrayDimensions = [] 
		MaxStringLength = 0
		IsOptional = 0
] 

Duration is a SubDataType of Double and explicit stated in the nodeset xml:

<Field DataType="Duration" Name="PlannedProductionTime">
  <Description>Planned production time</Description>
</Field>

But in the .bsd there is a the PlannedProductionTime a opc:Double

 <opc:StructuredType BaseType="ua:ExtensionObject" Name="JobListElementType">
  <opc:Documentation>Description of a job in a job list</opc:Documentation>
  <opc:Field TypeName="opc:CharArray" Name="JobName"/>
  <opc:Field TypeName="opc:CharArray" Name="JobDescription"/>
  <opc:Field TypeName="opc:CharArray" Name="JobClassification"/>
  <opc:Field TypeName="opc:CharArray" Name="CustomerName"/>
  <opc:Field TypeName="opc:CharArray" Name="ProductionDatasetName"/>
  <opc:Field TypeName="opc:CharArray" Name="ProductionDatasetDescription"/>
  <opc:Field TypeName="opc:Int32" Name="NoOfMaterial"/>
  <opc:Field LengthField="NoOfMaterial" TypeName="opc:CharArray" Name="Material"/>
  <opc:Field TypeName="opc:Int32" Name="NoOfProductName"/>
  <opc:Field LengthField="NoOfProductName" TypeName="opc:CharArray" Name="ProductName"/>
  <opc:Field TypeName="opc:Int32" Name="NoOfProductDescription"/>
  <opc:Field LengthField="NoOfProductDescription" TypeName="opc:CharArray" Name="ProductDescription"/>
  <opc:Field TypeName="opc:CharArray" Name="JobPriority"/>
  <opc:Field TypeName="opc:DateTime" Name="PlannedStart"/>
  **<opc:Field TypeName="opc:Double" Name="PlannedProductionTime"/>**
  <opc:Field TypeName="opc:DateTime" Name="LatestEnd"/>
 </opc:StructuredType>

From the encoding point of view there is no difference

@MartinaFelberFEM : What's your opinion?

MartinaFelberFEM, 2020-05-19:
Did you notice the following too? 

Id = ns=3;i=3024 DATATYPE
BrowseName = ns=3;ConfigurationParameterType
DisplayName = :ConfigurationParameterType

reference Server:
	[ 
		Name = Unit
		Description = :
		**DataType = i=0**
		ValueRank = -1
		ArrayDimensions = [] 
		MaxStringLength = 0
		IsOptional = 0
	] 

test server:
	[ 
		Name = Unit
		Description = :
		**DataType = i=887**
		ValueRank = -1
		ArrayDimensions = [] 
		MaxStringLength = 0
		IsOptional = 0
	] 

DataType i=0 at the reference server seems very odd to me.

I'll answer to your question asap.

MartinaFelberFEM, 2020-05-19:
Referring to **PlannedProductionTime**:

I have generated the .bsd file with an online base64 decoder, because Euromap does not provide a .bsd file.
I took the DataTypeDictionary bytestring of the XML Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml
There should not have been an error at base64 decoding, etc.

At first I thought that the .bsd maybe only allows built-in DataTypes.

But the DataType duration is described in the namespace 0 .bsd of OpcFoundation ...
https://github.com/OPCFoundation/UA-Nodeset/blob/master/Schema/Opc.Ua.Types.bsd

I don't know if this was intended by the Euromap committee.
Or they thought it does not matter because the en/decoding as you have already mentioned is the same.

I haven't found a lot information in the Opc specifications.

Part 3: 5.8.2 Encoding Rules for different kinds of DataTypes:
https://reference.opcfoundation.org/v104/Core/docs/Part3/5.8.2/

> Built-in DataTypes are a fixed set of DataTypes (see OPC 10000-6 for a complete list of Built-
in DataTypes). They have no encodings visible in the AddressSpace since the encoding should
be known to all OPC UA products. Examples of Built-in DataTypes are Int32 (see 8.26) and
Double (see 8.12).

> Simple DataTypes are subtypes of the Built-in DataTypes. They are handled on the wire like
the Built-in DataType, i.e. they cannot be distinguished on the wire from their Built-in
supertypes. Since they are handled like Built-in DataTypes regarding the encoding they cannot
have encodings defined in the AddressSpace. Clients can read the DataType Attribute of a
Variable or VariableType to identify the Simple DataType of the Value Attribute. An example of
a Simple DataType is Duration. It is handled on the wire as a Double but the Client can read
the DataType Attribute and thus interpret the value as defined by Duration (see 8.13).

I am not sure if the nodeset-compiler parses the <UADataType> tags from the XML or only uses the .bsd file for DataType information.

matkonnerth, 2020-05-20:
The nodesetCompiler uses the bsd for generating the types. When I import the xml, I take the information of the xml and skip the information of the base64 encoded bsd. This explains the difference.

matkonnerth, 2020-05-20:
I didn't notice the difference on datatype EM83;i=3024 - the nodesetcompiler generates this code for the members:

/* ConfigurationParameterType */
static UA_DataTypeMember ConfigurationParameterType_members[4] = {
{
    UA_TYPENAME("Id") /* .memberName */
    UA_TYPES_UINT32, /* .memberTypeIndex */
    0, /* .padding */
    true, /* .namespaceZero */
    false, /* .isArray */
    false /* .isOptional */
},
{
    UA_TYPENAME("Description") /* .memberName */
    UA_TYPES_LOCALIZEDTEXT, /* .memberTypeIndex */
    offsetof(UA_ConfigurationParameterType, description) - offsetof(UA_ConfigurationParameterType, id) - sizeof(UA_UInt32), /* .padding */
    true, /* .namespaceZero */
    false, /* .isArray */
    false /* .isOptional */
},
{
    UA_TYPENAME("DefaultValue") /* .memberName */
    UA_TYPES_VARIANT, /* .memberTypeIndex */
    offsetof(UA_ConfigurationParameterType, defaultValue) - offsetof(UA_ConfigurationParameterType, description) - sizeof(UA_LocalizedText), /* .padding */
    true, /* .namespaceZero */
    false, /* .isArray */
    false /* .isOptional */
},
{
    UA_TYPENAME("Unit") /* .memberName */
    UA_TYPES_EUINFORMATION, /* .memberTypeIndex */
    offsetof(UA_ConfigurationParameterType, unit) - offsetof(UA_ConfigurationParameterType, defaultValue) - sizeof(UA_Variant), /* .padding */
    false, /* .namespaceZero */
    false, /* .isArray */
    false /* .isOptional */
},};

The last member has typeIndex UA_TYPES_EUINFORMATION, this is correct, but the namespace zero flag is not correct, it should be namespace zero.

When I manually set the namespaceZero flag, the difference is gone. -> This is an nodesetCompiler issue.

MartinaFelberFEM, 2020-05-20:
Regarding EM83;i=3024 - EUInformation:
I'll try to fix it or open an issue at open62541 repository.

Regarding the different types from .bsd and xml (double vs. duration):

Probably both implementations (nodeset-compiler and NodesetLoader) are ok. 

Do you know if there are any nodeset XMLs which only use the DataTypeDictionary bytestring and do not provide the DataType information in the XML?

Regarding the integration test: 
I could manually edit the .bsd file so that the datatpyes match with the xml. There are 4 or 5 Duration/Double changes to replace I think ...
Then the tests would pass.
Would that be ok for you?

Otherwise I could probably search for a supertype (from Duration to Double) for DataType Definition attributes, but I am not sure if I can handle all cases. I don't think this is a good idea.

matkonnerth, 2020-05-20:
I've added here an issue and did a little debugging in the generator, here the issue https://github.com/open62541/open62541/issues/3618

matkonnerth, 2020-05-20:
hm, I don't know any nodeset, that only provides the encoded bsd. I think the recommended way is to provide it via the tags on the datatype.
I think you have to grab this open issue ;)
https://github.com/open62541/open62541/issues/2587

For me that would be ok to replace the Duration and Double.

MartinaFelberFEM, 2020-05-21:
Many thanks for opening the issue!

Replacing Double with Duration in the .bsd file unfortunately creates the same issues as with EUInformation datatype.
The namespaceZero flag in the generated referenceServer code is wrong and the DataType Id is 0  ...

```
{
    UA_TYPENAME("PlannedProductionTime") /* .memberName */
    UA_TYPES_DURATION, /* .memberTypeIndex */
    offsetof(UA_JobListElementType, plannedProductionTime) - offsetof(UA_JobListElementType, plannedStart) - sizeof(UA_DateTime), /* .padding */
    false, /* .namespaceZero */
    false, /* .isArray */
    false /* .isOptional */
},
```

matkonnerth, 2020-05-21:
okay, what if we change it in the xml from duration to double?

matkonnerth, 2020-05-21:
issue for nodesetcompiler is opened, I close this one.

closed 2020-05-21 by matkonnerth

MartinaFelberFEM, 2020-05-21:
> okay, what if we change it in the xml from duration to double?

Even if we would, we would still have the EUInformation datatype issue and the test would still fail.

I will observe the issue tomorrow on open62541 repo, maybe we can fix it. If not then I will rewrite the XML.