open62541/open62541-nodeset-loader #160: Too simple type names: TNode, TNodeId

gveector, 2021-08-02:
In NodesetLoader.h "TNode" / TNodeId.h "TNodeId" - is too simple type names, causes a type name conflict when using the nodesetLoader in projects.

Please change this internal names by adding a prefix or suffix to them. 
For example (NL - NodesetLoader):
TNode -> TNLNode / TNodeNL
TNodeId -> TNLNodeId / TNodeIdNL

Sorry to bother you.

matkonnerth, 2021-08-02:
I understand that, namespaces are missing in C ;)

I would vote for a _NL_ prefix for types, for example:
_NL_Node_
_NL_NodeId_

Most of the existing public functions are already prefixed with _NodesetLoader_, I will not change that.

matkonnerth, 2021-08-02:
@gveector : please take a look a this PR https://github.com/matkonnerth/nodesetLoader/pull/161

matkonnerth, 2021-08-02:
should be fixed with https://github.com/matkonnerth/nodesetLoader/pull/161

closed 2021-08-02 by matkonnerth

gveector, 2021-08-04:
Excellent!