open62541/open62541-nodeset-loader #163: Error while making

Hfiazihilscher, 2021-08-19:
Hello i am new in this topic and need some help. I work with the options:
ENABALE_CONAN ON
ENABLE_BACKEND_OPEN62541 ON

while making i recseive this error:

nodesetLoader/backends/open62541/src/DataTypeImporter.c:369:30: error: incompatible types when assigning to type ‘UA_UInt32’ {aka ‘unsigned int’} from type ‘UA_NodeId’ {aka ‘struct <anonymous>’}
  369 |     type->binaryEncodingId = getBinaryEncodingId(node);
         |                                              ^~~~~~~~~~~~~~~~~~~
nodesetLoader/backends/open62541/src/DataTypeImporter.c: In function ‘EnumDataType_init’:
/nodesetLoader/backends/open62541/src/DataTypeImporter.c:383:34:
 error: incompatible types when assigning to type ‘UA_UInt32’ {aka ‘unsigned int’} from type ‘UA_NodeId’ {aka ‘const struct <anonymous>’}
  383 |     enumType->binaryEncodingId = UA_NODEID_NULL;
         |                                                        ^~~~~~~~~~~~~~

Could the problem be: while making the compiler dont find the right Header-files? 

[image]

matkonnerth, 2021-08-19:
please disable conan usage and use the 1.2 branch of the open62541. The binary encoding id was changed from a integer to a NodeId.

matkonnerth, 2021-08-21:
I removed the open62541 library as conan dependency, because nodesetLoader needs version >= 1.2 of open26541, and this version is currently not available via conan: https://github.com/matkonnerth/nodesetLoader/pull/164

Hopefully you can upgrade to open62541 version >= 1.2 in your project, otherwise we have to figure out another solution.

Hfiazihilscher, 2021-08-23:
i already used this library Version 1.2. i will try it and bind the library with another way. Thanks a lot!

Hfiazihilscher, 2021-08-23:
[image]

matkonnerth, 2021-08-28:
it seems that you are using a master branch version of the open62541, here the UA_DataTypeMember from master:

https://github.com/open62541/open62541/blob/530e3c0aaffd07d85f260e1713de39bf2fd22181/include/open62541/types.h#L950

and here the one from v1.2.2
https://github.com/open62541/open62541/blob/ecf5a703785877a8719a0cda863a98455f7d5d12/include/open62541/types.h#L945

currently the nodesetLoader supports only the 1.2.2 version, but I am working on making it also work with the master branch:
https://github.com/matkonnerth/nodesetLoader/pull/165

Hfiazihilscher, 2021-08-30:
Thanks for your detailed help. After I installed the branch, I had the problem again which I posted last. I had to switch off the option: USE_MEMBERTYPE_INDEX. After that it compiled and I can now load my own nodesets!

closed 2021-08-30 by matkonnerth