open62541/open62541-nodeset-loader #178: "node with unresolved reference"

paddor, 2022-01-26:
With a nodeset XML file that works fine with another UA stack, I'm getting this error with the Open62541 backend:

```
[2022-01-26 23:41:34.528 (UTC+0100)] debug/userland     Start import nodeset: instances.xml
[2022-01-26 23:41:34.559 (UTC+0100)] error/userland     node with unresolved reference
[2022-01-26 23:41:34.559 (UTC+0100)] error/userland     importing the nodeset failed, nodes were not added
```

Is there a way to log more info? For example which node has an unresolved reference?

matkonnerth, 2022-01-27:
Is it possible for you to share the nodeset for debugging the issue?

paddor, 2022-01-27:
Yeah, I've just desensitized most identifiers using SHA1. [Here](https://gist.github.com/paddor/26bf30f99da07e3062a3431949a6d0e6)

I've also changed that error message so it includes the node ID of the offending node. It's simply the first `UAVariable` if I remember correctly.

matkonnerth, 2022-01-28:
ok, I can reproduce the issue. It seems the nodesetLoader cannot figure out, if the "HasCondition" reference is a hierachical or nonHierachical reference. 

When I build the open62541 from master with full namespace 0, I can import your nodeset without errors:

_2022-01-28 09:20:20.748 (UTC+0100)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
[2022-01-28 09:20:20.748 (UTC+0100)] info/server	AccessControl: Anonymous login is enabled
[2022-01-28 09:20:20.748 (UTC+0100)] info/server	AccessControl: x509 certificate user authentication is enabled
[2022-01-28 09:20:20.748 (UTC+0100)] warn/server	Username/Password Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2022-01-28 09:20:20.748 (UTC+0100)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2022-01-28 09:20:20.749 (UTC+0100)] debug/userland	Start import nodeset: ../nodesets/instances_sha1.xml
[2022-01-28 09:20:20.768 (UTC+0100)] debug/userland	imported ReferenceTypes: 0
[2022-01-28 09:20:20.768 (UTC+0100)] debug/userland	imported DataTypes: 0
[2022-01-28 09:20:20.768 (UTC+0100)] debug/userland	imported ObjectTypes: 0
[2022-01-28 09:20:20.774 (UTC+0100)] debug/userland	imported Objects: 935
[2022-01-28 09:20:20.775 (UTC+0100)] debug/userland	imported Methods: 176
[2022-01-28 09:20:20.775 (UTC+0100)] debug/userland	imported VariableTypes: 0
[2022-01-28 09:20:20.792 (UTC+0100)] debug/userland	imported Variables: 2557
[2022-01-28 09:20:20.792 (UTC+0100)] debug/userland	imported Views: 0
[2022-01-28 09:20:20.805 (UTC+0100)] info/eventloop	Starting the EventLoop
[2022-01-28 09:20:20.805 (UTC+0100)] info/network	TCP network layer listening on opc.tcp://debian:4840/_

matkonnerth, 2022-01-28:
the compiler switch for the open62541 with full namespace zero is _ UA_NAMESPACE_ZERO = FULL_ . Can you try it again with that option?

paddor, 2022-01-31:
That sounds promising, but unfortunately I can't seem to reproduce what you did. When I configure like this:

```
nodesetLoader/build  $ cmake -DCMAKE_BUILD_TYPE=Debug -DENABLE_BACKEND_OPEN62541=ON -DBUILD_SHARED_LIBS=ON -DUA_NAMESPACE_ZERO=FULL ..
```

Some of the last few lines are:

```-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    UA_NAMESPACE_ZERO
```

And when I compile, install, and compile my app and try to load the nodeset with it, it ends with the same error.

matkonnerth, 2022-01-31:
UA_NAMESPACE_ZERO is a variable of then open62541 project, you have to set it when you build the open.

paddor, 2022-01-31:
Oh, of course, my bad. It does a lot more now, but still ends in a SEGFAULT during import. Here's the output and the C level backtrace:

```
[2022-01-31 09:45:05.083 (UTC+0100)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2022-01-31 09:45:05.084 (UTC+0100)] info/server        AccessControl: Anonymous login is enabled
[2022-01-31 09:45:05.084 (UTC+0100)] info/server        AccessControl: x509 certificate user authentication is enabled
[2022-01-31 09:45:05.084 (UTC+0100)] warn/server        Username/Password Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2022-01-31 09:45:05.084 (UTC+0100)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2022-01-31 09:45:05.084 (UTC+0100)] debug/userland     Start import nodeset: lta_instances_sha1.xml
[2022-01-31 09:45:05.118 (UTC+0100)] debug/userland     imported ReferenceTypes: 0
[2022-01-31 09:45:05.118 (UTC+0100)] debug/userland     imported DataTypes: 0
[2022-01-31 09:45:05.118 (UTC+0100)] debug/userland     imported ObjectTypes: 0
[2022-01-31 09:45:05.118 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode: Node could not create a node with error code BadNodeAttributesInvalid
[2022-01-31 09:45:05.118 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode: Node could not create a node with error code BadNodeAttributesInvalid
[2022-01-31 09:45:05.118 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode: Node could not create a node with error code BadNodeAttributesInvalid
[2022-01-31 09:45:05.118 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode: Node could not create a node with error code BadNodeAttributesInvalid
[2022-01-31 09:45:05.118 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode: Node could not create a node with error code BadNodeAttributesInvalid
[2022-01-31 09:45:05.118 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode: Node could not create a node with error code BadNodeAttributesInvalid
[...]
[2022-01-31 09:45:05.131 (UTC+0100)] debug/userland     imported Objects: 935
[2022-01-31 09:45:05.131 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=2;s=35f3a68a42061ab6ef23369d1ea052d54a2c9509.StateCondition.AddComment): Parent node not found
[2022-01-31 09:45:05.131 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=2;s=35f3a68a42061ab6ef23369d1ea052d54a2c9509.StateCondition.AddComment): The parent reference for is invalid
[2022-01-31 09:45:05.131 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=2;s=35f3a68a42061ab6ef23369d1ea052d54a2c9509.StateCondition.Disable): Parent node not found
[2022-01-31 09:45:05.132 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=2;s=35f3a68a42061ab6ef23369d1ea052d54a2c9509.StateCondition.Disable): The parent reference for is invalid
[2022-01-31 09:45:05.132 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=2;s=35f3a68a42061ab6ef23369d1ea052d54a2c9509.StateCondition.Enable): Parent node not found
[2022-01-31 09:45:05.132 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=2;s=35f3a68a42061ab6ef23369d1ea052d54a2c9509.StateCondition.Enable): The parent reference for is invalid
[2022-01-31 09:45:05.132 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=2;s=35f3a68a42061ab6ef23369d1ea052d54a2c9509.StateCondition.Acknowledge): Parent node not found
[2022-01-31 09:45:05.132 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=2;s=35f3a68a42061ab6ef23369d1ea052d54a2c9509.StateCondition.Acknowledge): The parent reference for is invalid
[2022-01-31 09:45:05.132 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=2;s=5dbf87ed44691801f5d2269524f0ee6511c0e7fe.StateCondition.AddComment): Parent node not found
[2022-01-31 09:45:05.132 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=2;s=5dbf87ed44691801f5d2269524f0ee6511c0e7fe.StateCondition.AddComment): The parent reference for is invalid
[2022-01-31 09:45:05.132 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=2;s=5dbf87ed44691801f5d2269524f0ee6511c0e7fe.StateCondition.Disable): Parent node not found
[2022-01-31 09:45:05.132 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=2;s=5dbf87ed44691801f5d2269524f0ee6511c0e7fe.StateCondition.Disable): The parent reference for is invalid
[2022-01-31 09:45:05.132 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=2;s=5dbf87ed44691801f5d2269524f0ee6511c0e7fe.StateCondition.Enable): Parent node not found
[2022-01-31 09:45:05.132 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=2;s=5dbf87ed44691801f5d2269524f0ee6511c0e7fe.StateCondition.Enable): The parent reference for is invalid
[2022-01-31 09:45:05.132 (UTC+0100)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=2;s=5dbf87ed44691801f5d2269524f0ee6511c0e7fe.StateCondition.Acknowledge): Parent node not found
[... cut ...]
/home/user/.rubies/ruby-3.0.2/lib/libruby.so.3.0(rb_vm_exec+0xa64) [0x7f80e62d2cb4] vm.c:2172
/home/user/.rubies/ruby-3.0.2/lib/libruby.so.3.0(rb_ec_exec_node+0xed) [0x7f80e60f75ad] eval.c:317
/home/user/.rubies/ruby-3.0.2/lib/libruby.so.3.0(ruby_run_node+0x5a) [0x7f80e60fd55a] eval.c:375
/home/user/.rubies/ruby-3.0.2/bin/ruby(main+0x5f) [0x5559a65b317f] ./main.c:50
```

Is it possible that `isTrue(const char *s)` was called with a wrong argument?

My app so far just creates a server, gets its config, sets the port on the config, and then loads the XML nodeset.

paddor, 2022-02-01:
I made a minimal C program (no Ruby), compiled with LLVM and reproduced the error and backtrace in LLDB:

```
* thread #1, name = 'test', stop reason = signal SIGSEGV: invalid address (fault address: 0x1)
    frame #0: 0x00007ffff7680b7e libc.so.6`__strcmp_avx2 at strcmp-avx2.S:101
(lldb) bt
* thread #1, name = 'test', stop reason = signal SIGSEGV: invalid address (fault address: 0x1)
  * frame #0: 0x00007ffff7680b7e libc.so.6`__strcmp_avx2 at strcmp-avx2.S:101
    frame #1: 0x00007ffff7724cc6 libNodesetLoader.so`isTrue(s="") at conversion.h:34:9
    frame #2: 0x00007ffff772578f libNodesetLoader.so`setBoolean(adr=9768704, value="") at Value.c:28:26
    frame #3: 0x00007ffff77259e5 libNodesetLoader.so`setScalarValueWithAddress(adr=9768704, kind=0, value="") at Value.c:98:9
    frame #4: 0x00007ffff7725af2 libNodesetLoader.so`setPrimitiveValue(data=0x00000000009682a0, value="", kind=UA_DATATYPEKIND_BOOLEAN, memSize=0) at Value.c:137:5
    frame #5: 0x00007ffff77262c8 libNodesetLoader.so`setScalar(value=0x0000000000635c10, type=0x00007ffff7f9e5a0, data=0x00000000009682a0, customTypes=0x0000000000000000, serverContext=0x000000000061bdc0) at Value.c:324:9
    frame #6: 0x00007ffff77265c8 libNodesetLoader.so`Value_getData(value=0x0000000000635bb0, type=0x00007ffff7f9e5a0, customTypes=0x0000000000000000, serverContext=0x000000000061bdc0) at Value.c:401:9
    frame #7: 0x00007ffff77234bc libNodesetLoader.so`handleVariableNode(node=0x0000000000635a60, id=0x00007fffffffd8b0, parentId=0x00007fffffffd8f0, parentReferenceId=0x00007fffffffd8d0, lt=0x00007fffffffd870, qn=0x00007fffffffd850, description=0x00007fffffffd890, serverContext=0x000000000061bdc0) at import.c:255:16
    frame #8: 0x00007ffff7724195 libNodesetLoader.so`addNode(serverContext=0x000000000061bdc0, node=0x0000000000635a60) at import.c:416:9
    frame #9: 0x00007ffff7720281 libNodesetLoader.so`Nodeset_forEachNode(nodeset=0x00000000004cf770, nodeClass=NODECLASS_VARIABLE, context=0x000000000061bdc0, fn=(libNodesetLoader.so`addNode at import.c:374:1)) at Nodeset.c:631:9
    frame #10: 0x00007ffff771dff1 libNodesetLoader.so`NodesetLoader_forEachNode(loader=0x0000000000616c70, nodeClass=NODECLASS_VARIABLE, context=0x000000000061bdc0, fn=(libNodesetLoader.so`addNode at import.c:374:1)) at NodesetLoader.c:574:12
    frame #11: 0x00007ffff7724932 libNodesetLoader.so`addNodes(loader=0x0000000000616c70, serverContext=0x000000000061bdc0, logger=0x000000000063bc10) at import.c:552:13
    frame #12: 0x00007ffff7724b79 libNodesetLoader.so`NodesetLoader_loadFile(server=0x0000000000406910, path="lta_instances_sha1.xml", extensionHandling=0x0000000000000000) at import.c:606:9
    frame #13: 0x00000000004011fc test`main(argc=2, argv=0x00007fffffffdbd8) at test.c:25:18
    frame #14: 0x00007ffff75210b3 libc.so.6`__libc_start_main(main=(test`main at test.c:15), argc=2, argv=0x00007fffffffdbd8, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdbc8) at libc-start.c:308:16
    frame #15: 0x00000000004010ae test`_start + 46
```

matkonnerth, 2022-02-01:
hm, I cannot reproduce that, have tried your nodeset with gcc and clang. Here again the nodeset would be helpful, at least the node generating this segfault. 

Seems that _isTrue_ is called with an address of 0x1, where is this strange address coming from? Looks like uninitialized memory which is flagged with 0x1.

paddor, 2022-02-01:
I was trying with the same nodeset, with open62541 and nodesetLoader as shared libraries. I don't know where the 0x01 came from.

Now I just tried again with the two libraries statically linked and it works with no warnings or crashes. I'm confused.

matkonnerth, 2022-02-01:
with this nodeset provided [here](https://gist.github.com/paddor/26bf30f99da07e3062a3431949a6d0e6) ?

paddor, 2022-02-01:
Yes

paddor, 2022-02-01:
Finally it seems to work, even with Ruby. The 0x01 earlier might have been because of a missing `-DUSE_MEMBERTYPE_INDEX=OF`. And the library order (`-lNodesetLoader -lopen62541 -lxml2`) really matters. I learned a lot today. 😅

closed 2022-02-01 by paddor