open62541/open62541-nodeset-loader #180: error: conversion to ‘UA_Int16’ {aka ‘short int’} from ‘short unsigned int’ may change the sign of the result

paddor, 2022-02-09:
I just upgraded from Clang 10.0.0 to Clang 13.0.1 and now I'm getting these errors when compiling NodesetLoader:

```
In file included from /home/roadster/dev/oss/open62541-ruby/ext/open62541/deps/nodesetLoader/backends/open62541/src/import.c:11:
/home/roadster/dev/oss/open62541-ruby/ext/open62541/deps/nodesetLoader/backends/open62541/src/conversion.h: In function ‘UA_DateTime_fromString’:
/home/roadster/dev/oss/open62541-ruby/ext/open62541/deps/nodesetLoader/backends/open62541/src/conversion.h:85:15: error: conversion to ‘UA_Int16’ {aka ‘short int’} from ‘short unsigned int’ may change the sign of the result [-Werror=sign-conversion]
   85 |     dt.year = (UA_UInt16)y;
      |               ^
In file included from /home/roadster/dev/oss/open62541-ruby/ext/open62541/deps/nodesetLoader/backends/open62541/src/DataTypeImporter.c:9:
/home/roadster/dev/oss/open62541-ruby/ext/open62541/deps/nodesetLoader/backends/open62541/src/conversion.h: In function ‘UA_DateTime_fromString’:
/home/roadster/dev/oss/open62541-ruby/ext/open62541/deps/nodesetLoader/backends/open62541/src/conversion.h:85:15: error: conversion to ‘UA_Int16’ {aka ‘short int’} from ‘short unsigned int’ may change the sign of the result [-Werror=sign-conversion]
   85 |     dt.year = (UA_UInt16)y;
      |               ^
[ 86%] Building C object CMakeFiles/NodesetLoader.dir/backends/open62541/src/customDataType.c.o
In file included from /home/roadster/dev/oss/open62541-ruby/ext/open62541/deps/nodesetLoader/backends/open62541/src/Value.c:9:
/home/roadster/dev/oss/open62541-ruby/ext/open62541/deps/nodesetLoader/backends/open62541/src/conversion.h: In function ‘UA_DateTime_fromString’:
/home/roadster/dev/oss/open62541-ruby/ext/open62541/deps/nodesetLoader/backends/open62541/src/conversion.h:85:15: error: conversion to ‘UA_Int16’ {aka ‘short int’} from ‘short unsigned int’ may change the sign of the result [-Werror=sign-conversion]
   85 |     dt.year = (UA_UInt16)y;
```

I can turn it into a warning with `-Wno-error=sign-conversion`, but maybe there's a cleaner way to avoid this error/warning.

paddor, 2022-02-09:
This is actually happening with GCC. I don't know why this error just started popping up today after I upgraded Clang.

```
/usr/bin/cmake --target NodesetLoader -DENABLE_BACKEND_OPEN62541=ON -DCMAKE_INSTALL_PREFIX=/home/user/dev/oss/open62541-ruby/tmp/x86_64-linux/open62541/3.0.3/stage -DUSE_MEMBERTYPE_INDEX=OFF -DCMAKE_BUILD_TYPE=Debug -DCMAKE_C_FLAGS_DEBUG='-fPIC
 -Wno-error=sign-conversion -O0 -g' -DCMAKE_INTERPROCEDURAL_OPTIMIZATION=OFF /home/user/dev/oss/open62541-ruby/ext/open62541/deps/nodesetLoader
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Downloading conan.cmake from https://github.com/conan-io/cmake-conan
-- Conan: Automatic detection of conan settings from cmake
-- Conan: Settings= -s;build_type=Debug;-s;compiler=gcc;-s;compiler.version=9;-s;compiler.libcxx=libstdc++11
-- Conan: checking conan executable
-- Conan: Found program /usr/bin/conan
-- Conan: Version found Conan version 1.44.1
```

matkonnerth, 2022-02-11:
seems the open62541 changed the the year member of the struct from short unsigned integer to short integer. I try to prepare a fix.

matkonnerth, 2022-02-11:
here the PR with the fix: https://github.com/matkonnerth/nodesetLoader/pull/182

closed 2022-02-14 by matkonnerth