open62541/open62541-nodeset-loader #186: Import of Union DataTypes is broken

MurzynJ, 2022-02-16:
Padding is not calculated correctly when creating the data type structures.

See example definitions:
[image]

And the resulting structures generated by open62541's NodesetCompiler, and by the NodesetLoader:
[image]

For a reference I'm attaching the sample XML (with .txt extension as Github won't allow .xml):
[issues.xml.txt](https://github.com/matkonnerth/nodesetLoader/files/8077821/issues.xml.txt)

matkonnerth, 2022-09-16:
Hi @MurzynJ , here is a fix for this issue: https://github.com/open62541/nodesetLoader/pull/192

jpfr, 2026-09-05:
The original union layout and padding fix was merged previously. Current union construction and imported-value coverage landed with open62541/open62541#8381, and the core XML union-value decoder is also being backported to open62541 1.5 in open62541/open62541#8383. Closing as fixed.

closed 2026-09-05 by jpfr