open62541/open62541-nodeset-loader #189: Unable to load the FX nodeset files in nodesetloader

rushabhi, 2022-04-19:
Hi, 
I am trying to load FX nodeset files, but I am getting the following issue. If you can help with this. 

debian@debian:~/nodesetLoader/build/bin$ ./server FX.xml 
[2022-04-19 01:14:55.404 (UTC-0400)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
[2022-04-19 01:14:55.404 (UTC-0400)] info/server	AccessControl: Anonymous login is enabled
[2022-04-19 01:14:55.404 (UTC-0400)] info/server	AccessControl: x509 certificate user authentication is enabled
[2022-04-19 01:14:55.404 (UTC-0400)] warn/server	Username/Password Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2022-04-19 01:14:55.404 (UTC-0400)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2022-04-19 01:14:55.404 (UTC-0400)] debug/userland	Start import nodeset: FX.xml
[2022-04-19 01:14:55.415 (UTC-0400)] debug/userland	imported ReferenceTypes: 25
Illegal instruction

matkonnerth, 2022-05-23:
Hi @rushabhi, can you provide the nodeset for debugging/testing?

matkonnerth, 2023-06-25:
tested without errors:
Opc.Ua.Di.NodeSet2.xml
UAFX/opc.ua.fx.data.nodeset2.xml
UAFX/opc.ua.fx.data.nodeset2.xml

please open a new issue, if the error persists.

closed 2023-06-25 by matkonnerth