open62541/open62541-nodeset-loader #201: Unable to reference nodes from dependency companion nodesets

keba-uso, 2022-10-05:
This issue appeared when testing the **nodesetLoader** with open62541 support, with the **UA-Nodesets** companion specifications:

- Repository: https://github.com/OPCFoundation/UA-Nodeset
- Branch: [latest](https://github.com/OPCFoundation/UA-Nodeset/tree/latest)
- Commit: [a2208e8](https://github.com/OPCFoundation/UA-Nodeset/commit/a2208e8960453aabbfeacee94aee287292398f3a)

The companion specifications targeted by this issue are the following:

- [FDI5](https://github.com/OPCFoundation/UA-Nodeset/blob/latest/FDI/Opc.Ua.Fdi5.NodeSet2.xml)
- [POWERLINK](https://github.com/OPCFoundation/UA-Nodeset/blob/latest/POWERLINK/Opc.Ua.POWERLINK.NodeSet2.xml)

For example, when trying to load the FDI5 nodeset, the open62541 provides following issue:

- **_AddNode (ns=3;i=10): Parent node not found_**

This node and its parent nodes could not be added to the OPC UA address space. Here is an example node from `Opc.Ua.Fdi5.NodeSet2.xml`:

\<UAVariableType NodeId="ns=1;i=2" BrowseName="1:UIPlugInType" DataType="Byte" ValueRank="1" ArrayDimensions="0"\>
&emsp;&emsp;\<DisplayName>UIPlugInType\</DisplayName\>
&emsp;&emsp;\<References\>
&emsp;&emsp;&emsp;&emsp;\<Reference ReferenceType="HasProperty"\>ns=1;i=3\</Reference\>
&emsp;&emsp;&emsp;&emsp;\<Reference ReferenceType="HasProperty"\>ns=1;i=4\</Reference\>
&emsp;&emsp;&emsp;&emsp;\<Reference ReferenceType="HasProperty"\>ns=1;i=5\</Reference\>
&emsp;&emsp;&emsp;&emsp;\<Reference ReferenceType="HasProperty"\>ns=1;i=6\</Reference\>
&emsp;&emsp;&emsp;&emsp;\<Reference ReferenceType="HasProperty"\>ns=1;i=7\</Reference\>
&emsp;&emsp;&emsp;&emsp;\<Reference ReferenceType="HasProperty"\>ns=1;i=8\</Reference\>
&emsp;&emsp;&emsp;&emsp;\<Reference ReferenceType="HasProperty"\>ns=1;i=9\</Reference\>
&emsp;&emsp;&emsp;&emsp;\<Reference ReferenceType="HasComponent">ns=1;i=10\</Reference\>
&emsp;&emsp;&emsp;&emsp;**\<Reference ReferenceType="HasSubtype" IsForward="false"\>ns=2;i=6246\</Reference\>**
&emsp;&emsp;\</References\>
\</UAVariableType\>

After investigating this issue, found out that the Node with id `ns=2;i=6246`, coming from the [DI](https://github.com/OPCFoundation/UA-Nodeset/blob/latest/DI/Opc.Ua.Di.NodeSet2.xml) nodeset, causes this error. The same issue appeared within the POWERLINK nodeset, since it also has nodes from dependent nodesets.

For some reason, when searching for references, the dependency nodesets are not included. Resolution of this issue may also target changes within the [open62541](https://github.com/open62541/open62541) repository.

matkonnerth, 2023-02-06:
you have to load the DI nodeset and then the FDI, then the "UIPluginType" is loaded. The real cause of the error is that the node with NodeId(3,10) is an object, which is referenced with "HasComponent" from an VariableType. Don't think this is valid opc ua modelling.

From a technical view here the object nodes are getting inserted before the variable types, so in our case the parent(VariableType) is not availabe when inserting the child(Object): https://github.com/open62541/open62541-nodeset-loader/blob/a4a7ce087fed6d6fa9a2af7f21bdfc3fa01bace6/backends/open62541/src/import.c#L546

matkonnerth, 2023-06-25:
I will close this issue, the following issues cannot be solved by the nodesetloader, are inside the Opc.Ua.AMLBaseTypes.NodeSet2.xml

- Method NodeId="ns=1;i=92" has no parent reference
- UAObject NodeId="ns=1;i=182" has an abstract ObjectType specified

closed 2023-06-25 by matkonnerth