open62541/open62541-nodeset-loader #205: Missing inverse references in companion specification

keba-uso, 2022-10-17:
This issue is related to the loading of companion nodeset files into the OPC UA server (using [open62541](https://github.com/open62541/open62541)). The version of the UA-Nodeset tested:

- Repository: **https://github.com/OPCFoundation/UA-Nodeset**
- Branch: **latest**
- Commit: **a2208e8**

When nodes in a nodeset are missing inverse references to parent nodes, the following issues appear when trying to load the nodeset to the OPC UA Server:

- _AddNode (NODE_ID): Parent node not found_
- _AddNode (NODE_ID): Reference type to the parent not found_

The following UA-Nodes specifications have these issues:
- [AMLBaseTypes](https://github.com/OPCFoundation/UA-Nodeset/blob/latest/AML/Opc.Ua.AMLBaseTypes.NodeSet2.xml)
- [FDI5](https://github.com/OPCFoundation/UA-Nodeset/blob/latest/FDI/Opc.Ua.Fdi5.NodeSet2.xml)
- [FDI7](https://github.com/OPCFoundation/UA-Nodeset/blob/latest/FDI/Opc.Ua.Fdi7.NodeSet2.xml)
- [IEC61850-6](https://github.com/OPCFoundation/UA-Nodeset/blob/latest/IEC61850/Opc.Ua.IEC61850-6.NodeSet2.xml)
- [ISA95](https://github.com/OPCFoundation/UA-Nodeset/blob/latest/ISA-95/Opc.ISA95.NodeSet2.xml)
- [MTConnect](https://github.com/OPCFoundation/UA-Nodeset/blob/latest/MTConnect/Opc.Ua.MTConnect.NodeSet2.xml)
- [POWERLINK](https://github.com/OPCFoundation/UA-Nodeset/blob/latest/POWERLINK/Opc.Ua.POWERLINK.NodeSet2.xml)

While the nodesetLoader loads the nodeset without any critical errors, the problem is that the issued nodes are not browsable within the OPC UA address space (screenshot when testing the server with UAExpert):

[image]

The open62541 Pythod-based nodeset compiler handles these issues by assuming the reverse direction and always adding the inverse relation: [nodeset_compiler.py:176](https://github.com/open62541/open62541/blob/de7534e3573277fa3dde511044c28754e272f1a3/tools/nodeset_compiler/nodeset_compiler.py#L176). We should consider adding the same functionality within the nodesetLoader as well. Then these issues should be resolved.

matkonnerth, 2023-02-06:
I've taken a look at the AMLBaseTypes. There is a ParentNodeId specified, but no reference type.
[ParentNodeId, but no reference type](https://github.com/OPCFoundation/UA-Nodeset/blob/3f301f123c3df4b1dfcafbcceff4ae756290c444/AML/Opc.Ua.AMLBaseTypes.NodeSet2.xml#L182)
in contrast to this variable will be added [ParentNodeId, and reference Type](https://github.com/OPCFoundation/UA-Nodeset/blob/3f301f123c3df4b1dfcafbcceff4ae756290c444/AML/Opc.Ua.AMLBaseTypes.NodeSet2.xml#L189)

From my point of view this is an issue in the AML nodeset.

matkonnerth, 2023-02-06:
here is the code section with is considering the ParentNodeId: https://github.com/open62541/open62541-nodeset-loader/blob/3760a45d2706b83bc91503d1e40f04394bc3f2ec/src/Sort.c#L300

juwalter, 2025-07-06:
also, (at least) in https://files.opcfoundation.org/schemas/UA/1.04/UANodeSet.xsd, `ParentNodeId` seems optional (missing ` use="required"`)

```xml
<xs:complexType name="UAInstance">
    <xs:complexContent>
        <xs:extension base="UANode">
            <xs:attribute name="ParentNodeId" type="NodeId"/>
        </xs:extension>
    </xs:complexContent>
</xs:complexType>
```

can and should [open62541-nodeset-loader](https://github.com/open62541/open62541-nodeset-loader/blob/master/src/Sort.c#L306) even rely on `ParentNodeId` at all? since it is optional and seemingly intended for design tools (as opposed to constructing the (authoritative) truth/address space)?

additional opinions:

- https://opcfoundation.org/forum/opc-ua-standard/parentnodeid-and-references/#p2235
  - source (end of page): https://reference.opcfoundation.org/Core/Part6/v104/docs/F.7
    > The NodeId of the Node that is the parent of the Node within the information model. This field is used to indicate that a tight coupling exists between the Node and its parent (e.g. when the parent is deleted the child is deleted as well). This information does not appear in the AddressSpace and is intended for use by design tools.
- https://github.com/OPCFoundation/UA-Nodeset/issues/60#issuecomment-640907962
  - source (end of page): https://reference.opcfoundation.org/v104/Core/docs/Part6/F.3/
    > A UANodeSet is expected to contain many UANodes which reference each other. Tools that create UANodeSets should not add Reference elements for both directions in order to minimize the size of the XML file. Tools that read the UANodeSets shall automatically add reverse references unless reverse references are not appropriate given the ReferenceType semantics. 

according to spec then: 

> ... tools that read the UANodeSets shall automatically add reverse references ...

jpfr, 2026-09-05:
The new importer resolves parent relationships from references declared in either direction and applies reverse relations through the server. The related core validation fix accepting all hierarchical parent ReferenceTypes is being backported to 1.5 in open62541/open62541#8383. Complete reference-graph resolution landed on master through open62541/open62541#8381. Closing as fixed for current master.

closed 2026-09-05 by jpfr