open62541/open62541-nodeset-loader #207: Export OPC-UA to XML

vvpnet, 2022-11-01:
Hello. Tell me, will there be an export to an XML file from a running OPC UA server? Or maybe there are already solutions?

xydan83, 2022-12-21:
This is a good question. I want to know it too.

matkonnerth, 2023-01-22:
This library supports importing nodeset files into the server address space, there is currently no support for exporting. I think the UAExpert has the ability to export a namespace from a connected server.

But let me know, what is the use case behind where you need to export the namespace from a running Server?

vvpnet, 2023-01-23:
Hi. The export function is very necessary. In order to be able to save all nodes created using OPC UA clients, for example, after the server shutdown and not only. And in order to restore nodes when starting the server, for example, from a database of time series data.

vvpnet, 2023-01-24:
> If you already have a database, you could store your business objects in there and restore them from there. That’s how we do it.

With the restoration, everything is clear! And what about the entry of nodes in the database? Clients create new nodes on the OPC UA server - these nodes (node structure) need to be saved to the database. Is there a solution to how this can be done?