open62541/open62541-nodeset-loader #217: handleMethodNode dont use arguments

allreadyexists, 2022-12-10:
I try to use nodesetLoader for load xml file contains Method with Arguments, but client don't see Arguments. In source code i see that method ignore Arguments(set it is NULL):

static void
handleMethodNode(const NL_MethodNode *node, UA_NodeId *id,
                 const UA_NodeId *parentId, const UA_NodeId *parentReferenceId,
                 const UA_LocalizedText *lt, const UA_QualifiedName *qn,
                 const UA_LocalizedText *description, UA_Server *server)
{
    UA_MethodAttributes attr = UA_MethodAttributes_default;
    attr.executable = isValTrue(node->executable);
    attr.userExecutable = isValTrue(node->userExecutable);
    attr.displayName = *lt;
    attr.description = *description;

    UA_Server_addMethodNode(server, *id, *parentId, *parentReferenceId, *qn,
                            attr, NULL, 0, NULL, 0, NULL, node->extension,
                            NULL);
}

Can you fix it?

matkonnerth, 2023-01-22:
please propose a PR, thank you

matkonnerth, 2023-01-28:
Input and output arguments are modelled with properties (variables with hasProperty reference) below the method. If these properties are in the nodeset, the should get imported and the client can read this properties for getting the args.

Please check your xml file for these properties.

closed 2023-12-08 by matkonnerth