open62541/open62541-nodeset-loader #219: New release

xydan83, 2022-12-21:
Hi!
There are a lot of new updates in the master branch, maybe someone knows when the new release will be released?

matkonnerth, 2023-02-02:
always live at head ;)

vvpnet, 2024-01-19:
Hello! How are things going with export in xml? I take it something is not working out for you? Or did you just change your mind about the open publication of the project?
@xydan83

xydan83, 2024-01-25:
> Hello! How are things going with export in xml? I take it something is not working out for you? Or did you just change your mind about the open publication of the project? @xydan83

The first version of the exporter is ready. I posted it [here](https://github.com/xydan83/open62541-nodeset-exporter) and there is also a DEMO on how to use it.
In the future, I hope to add it to the general **Open62541** project, similar to the **opent62541-nodeset-loader**. Together they will complete the basic interfaces of the export-import nodes.
But my exporter has a feature. I wrote it in C++, so I will need to think about an adapter for C so that I can easily connect it as a plugin to the library. I also use the 20th standards, it might be worth considering the option of redoing it a little and downgrading to the 11th standard to cover a larger number of older compilers for assembly.
But despite this, it can already be used.

vvpnet, 2024-01-30:
@xydan83 
Hello! I really liked your plans for the development of this project! I especially like using the plugin on the server side. In this mode, you can implement automatic data export when the ua tree structure changes. And it should work faster than on the client side. For me, the ultimate goal is to fully support automatic export to an xml file for embedded devices. And for large export/import servers to the POSTGRESQL database (While there are thoughts to use the XML plugin in PostgreSQL).

xydan83, 2024-01-30:
> @xydan83 Hello! I really liked your plans for the development of this project! I especially like using the plugin on the server side. In this mode, you can implement automatic data export when the [...]

Hi!
Thanks, I'm glad to read this)
Yes, from the server side, of course, export will be faster; for this you need to implement an interface for the server with similar methods as for the client.
The speed on the client also depends on what algorithm you use to collect the nodes needed for export.