open62541/open62541-nodeset-loader #224: memleak when using VariableNodes with string-NodeId as value

Fischer-Thomas, 2023-02-09:
Memory is allocated for [string-nodeIds](https://github.com/open62541/open62541-nodeset-loader/blob/master/backends/open62541/src/Value.c#L208-L220) but the pointer isn't stored. 

This happens for both scalar and array VariableNodes with DataType NodeId and the NodeIdentifier is a string.
This is also a problem when having structured DataTypes that have a NodeId as member (e.g. UA_PublishedVariableDataType).

see valgrind output below for attached example NodeSet File:
[memleak_stringnodeids.zip](https://github.com/open62541/open62541-nodeset-loader/files/10694994/memleak_stringnodeids.zip)

> ==320953==
> ==320953== HEAP SUMMARY:
> ==320953==     in use at exit: 51 bytes in 5 blocks
> ==320953==   total heap usage: 151,168 allocs, 151,163 frees, 89,772,609 bytes allocated
> ==320953==
> ==320953== 3 bytes in 1 blocks are definitely lost in loss record 1 of 3
> ==320953==    at 0x483AB65: calloc (vg_replace_malloc.c:760)
> ==320953==    by 0x70C155: UA_Array_copy (ua_types.c:1968)
> ==320953==    by 0x706FE7: String_copy (ua_types.c:165)
> ==320953==    by 0x70A5BB: UA_copy (ua_types.c:1407)
> ==320953==    by 0x7296B1: UA_String_copy.lto_priv.11 (types_generated_handling.h:373)
> ==320953==    by 0x729B2D: parse_nodeid_body (ua_types_lex.c:105)
> ==320953==    by 0x729DAC: parse_nodeid (ua_types_lex.c:186)
> ==320953==    by 0x729F49: UA_NodeId_parse (ua_types_lex.c:259)
> ==320953==    by 0x73DF73: extractNodeId (conversion.h:53)
> ==320953==    by 0x73F030: setNodeId (Value.c:212)
> ==320953==    by 0x73F716: setScalar (Value.c:372)
> ==320953==    by 0x73FA37: Value_getData (Value.c:450)
> ==320953==    by 0x73571D: handleVariableNode (import.c:258)
> ==320953==    by 0x73641B: addNodeImpl (import.c:419)
> ==320953==    by 0x74320F: Nodeset_forEachNode (Nodeset.c:629)
> ==320953==
> ==320953== 18 bytes in 1 blocks are definitely lost in loss record 2 of 3
> ==320953==    at 0x483AB65: calloc (vg_replace_malloc.c:760)
> ==320953==    by 0x70C155: UA_Array_copy (ua_types.c:1968)
> ==320953==    by 0x706FE7: String_copy (ua_types.c:165)
> ==320953==    by 0x70A5BB: UA_copy (ua_types.c:1407)
> ==320953==    by 0x7296B1: UA_String_copy.lto_priv.11 (types_generated_handling.h:373)
> ==320953==    by 0x729B2D: parse_nodeid_body (ua_types_lex.c:105)
> ==320953==    by 0x729DAC: parse_nodeid (ua_types_lex.c:186)
> ==320953==    by 0x729F49: UA_NodeId_parse (ua_types_lex.c:259)
> ==320953==    by 0x73DF73: extractNodeId (conversion.h:53)
> ==320953==    by 0x73F030: setNodeId (Value.c:212)
> ==320953==    by 0x73F716: setScalar (Value.c:372)
> ==320953==    by 0x73F40A: setStructure (Value.c:295)
> ==320953==    by 0x73F807: setScalar (Value.c:394)
> ==320953==    by 0x73FA37: Value_getData (Value.c:450)
> ==320953==    by 0x73571D: handleVariableNode (import.c:258)
> ==320953==
> ==320953== 30 bytes in 3 blocks are definitely lost in loss record 3 of 3
> ==320953==    at 0x483AB65: calloc (vg_replace_malloc.c:760)
> ==320953==    by 0x70C155: UA_Array_copy (ua_types.c:1968)
> ==320953==    by 0x706FE7: String_copy (ua_types.c:165)
> ==320953==    by 0x70A5BB: UA_copy (ua_types.c:1407)
> ==320953==    by 0x7296B1: UA_String_copy.lto_priv.11 (types_generated_handling.h:373)
> ==320953==    by 0x729B2D: parse_nodeid_body (ua_types_lex.c:105)
> ==320953==    by 0x729DAC: parse_nodeid (ua_types_lex.c:186)
> ==320953==    by 0x729F49: UA_NodeId_parse (ua_types_lex.c:259)
> ==320953==    by 0x73DF73: extractNodeId (conversion.h:53)
> ==320953==    by 0x73F030: setNodeId (Value.c:212)
> ==320953==    by 0x73F716: setScalar (Value.c:372)
> ==320953==    by 0x73F935: setArray (Value.c:422)
> ==320953==    by 0x73F9F0: Value_getData (Value.c:444)
> ==320953==    by 0x73571D: handleVariableNode (import.c:258)
> ==320953==    by 0x73641B: addNodeImpl (import.c:419)
> ==320953==
> ==320953== LEAK SUMMARY:
> ==320953==    definitely lost: 51 bytes in 5 blocks
> ==320953==    indirectly lost: 0 bytes in 0 blocks
> ==320953==      possibly lost: 0 bytes in 0 blocks
> ==320953==    still reachable: 0 bytes in 0 blocks
> ==320953==         suppressed: 0 bytes in 0 blocks
> ==320953==
> ==320953== For lists of detected and suppressed errors, rerun with: -s
> ==320953== ERROR SUMMARY: 3 errors from 3 contexts (suppressed: 0 from 0)

@andreasebner

matkonnerth, 2023-06-03:
thank you for the test nodeset, I can reproduce it, here is the PR fixing it: https://github.com/open62541/open62541-nodeset-loader/pull/229

matkonnerth, 2023-06-03:
fixed with https://github.com/open62541/open62541-nodeset-loader/pull/229

closed 2023-06-03 by matkonnerth