open62541/open62541-nodeset-loader #231: Problem with String and ByteString in NodeID.

xydan83, 2023-06-23:
Hi!
I found a problem with String and ByteString in Node ID when I was trying to import a simple xml.

```xml
<?xml version="1.0" encoding="UTF-8"?>
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd">
    <NamespaceUris>
        <Uri>urn:rozhkov-nb:UnifiedAutomation:UaServerCpp</Uri>
        <Uri>http://www.unifiedautomation.com/DemoServer/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
        <Alias Alias="Organizes">i=35</Alias>
        <Alias Alias="UInt32">i=7</Alias>
    </Aliases>
    <UAObject NodeId="ns=2;s=Demo.Static" BrowseName="2:Static" ParentNodeId="i=85">
        <DisplayName>000_Static</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=61</Reference>
            <Reference ReferenceType="Organizes">ns=2;i=4294967294</Reference>
            <Reference ReferenceType="Organizes" IsForward="false">i=85</Reference>
        </References>
    </UAObject>
    <UAVariable NodeId="ns=2;i=4294967294" BrowseName="2:VariableWithNumericNodeId" ParentNodeId="ns=2;s=Demo.Static" DataType="UInt32" AccessLevel="3" UserAccessLevel="3">
        <DisplayName>VariableWithNumericNodeId</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
            <Reference ReferenceType="Organizes" IsForward="false">ns=2;s=Demo.Static</Reference>
        </References>
    </UAVariable>
</UANodeSet>
```

An error message appears during the import process.
[image]

The trace tells us that the problem may be related to an attempt to remove some node container.
[image]

When I replace String Node ID with Index everything is fine. The import completes successfully.
By the way, ByteString has the same problem...

xydan83, 2023-06-23:
I think the problem might be in the string reference with the second Node.

[image]

paddor, 2023-06-23:
Possibly related to https://github.com/open62541/open62541/issues/5794 ?

matkonnerth, 2023-06-23:
looks like this issue is related to: https://github.com/open62541/open62541/issues/5794

xydan83, 2023-06-23:
Yes, it seems. But I can't find anything in the standard about the mandatory order of nodes in xml.

matkonnerth, 2023-06-23:
there is no required order in the xml, this is definitive a bug within the nodesetloader

xydan83, 2023-06-23:
[image]

Very strange. How it possibly?

[image]

matkonnerth, 2023-06-24:
no deep copy of String copy was made in sort.c, here is the fix https://github.com/open62541/open62541-nodeset-loader/pull/232/commits

closed 2023-06-27 by matkonnerth