open62541/open62541-nodeset-loader #245: support for MinimumSamplingInterval missing

matkonnerth, 2023-11-28:


piperoc, 2024-03-23:
I added and tested the _MinimumSamplingAttribute_ for _UAVariable_.

You can check my modifications in my fork [piperoc/open62541-nodeset-loader](https://github.com/piperoc/open62541-nodeset-loader) . Look for the branch `add-minSamplingInterval` .

not sure you wanted me to use PR's or so, therefore I figured I'd fork it.

Please double check: I tested with a few nodesets. Some nodesets give me troubles in general (for example, the euromap ones are weird for me).

matkonnerth, 2024-03-26:
thank you, I've create a PR from your work. 

Which euromap nodesets generate troubles?

matkonnerth, 2024-03-26:
looks good, merged it to master.

piperoc, 2024-03-26:
> thank you, I've create a PR from your work.
> 
> Which euromap nodesets generate troubles?

Thank you @matkonnerth . I haven't tested extensively, but here's an error I get all the time with the following:

[image]

Could it be due to the UAModeler extensions?

matkonnerth, 2024-03-26:
seems that there is a circle with hierachical references between some nodes (node1 -> node2 ->node1), can you provide me the nodeset?

piperoc, 2024-03-26:
Sure @matkonnerth, the nodeset I used is the one in `open62541-nodeset-loader/nodesets/euromap_instances/euromapinstances.xml` .

jpfr, 2026-09-05:
MinimumSamplingInterval support was already merged into this repository and was retained in the loader integrated by open62541/open62541#8381. This issue can be closed.

closed 2026-09-05 by jpfr