open62541/open62541-nodeset-loader #253: Custom/Aliased UADataTypes and UAVariableTypes not supported

paddor, 2024-03-15:
I noticed that `UA_findDataTypeWithCustom()` can't find the data types of imported UAVariables with custom/aliased UADataTypes.

Given the following namespaces:
```
  <NamespaceUris>
    <Uri>http://foobar.example.com/Types/v2</Uri>
    <Uri>http://foobar.example.co/Instances</Uri>
  </NamespaceUris>
```

Given these Aliases:
```xml
    <Alias Alias="DataItemType">i=2365</Alias>
    <Alias Alias="AlarmStateStates">ns=1;s=af28ed20-dbb6-5624-aa65-4b42023ab1c4</Alias>
    <Alias Alias="AlarmStateType">ns=1;s=0cde5300-0cef-4699-b5c3-acb5011e0aa3</Alias>
```

And this UADataType:
```xml
  <UADataType NodeId="ns=1;s=af28ed20-dbb6-5624-aa65-4b42023ab1c4" BrowseName="1:AlarmStateStates">
    <DisplayName>AlarmStateStates</DisplayName>
    <Description Locale="de">Alarm- und Fehlerzustände</Description>
    <References>
      <Reference ReferenceType="HasProperty">ns=1;s=91428b80-ccb1-5adb-8685-5d51c867a0d3</Reference>
      <Reference ReferenceType="HasSubtype" IsForward="false">Enumeration</Reference>
    </References>
    <Definition Name="1:AlarmStateStates">
      <Field Name="Inaktiv" Value="0"/>
      <Field Name="Aktiv" Value="1"/>
    </Definition>
  </UADataType>
```

And this `UAVariableType`:
```xml
  <UAVariableType NodeId="ns=1;s=0cde5300-0cef-4699-b5c3-acb5011e0aa3" BrowseName="1:AlarmStateType" DataType="AlarmStateStates">
    <DisplayName>AlarmStateType</DisplayName>
    <Description Locale="de">Alarm- und Fehlerzustände</Description>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">DataItemType</Reference>
    </References>
  </UAVariableType>
```

And this `UAVariable`:
```xml
  <UAVariable NodeId="ns=2;g=16001624-6a1d-42d5-81d1-f54af84f5325" BrowseName="1:E_ComState" ParentNodeId="ns=2;g=33b2b39b-57a5-4d9f-af09-ae2f00ae1776" DataType="AlarmStateStates" AccessLevel="1">
    <DisplayName>E_ComState</DisplayName>
    <Description Locale="de">Energiemessung Messzelle Z.X Fehler: Kommunikation</Description>
    <References>
      <Reference ReferenceType="HasComponent" IsForward="false">ns=2;g=33b2b39b-57a5-4d9f-af09-ae2f00ae1776</Reference>
      <Reference ReferenceType="HasTypeDefinition">AlarmStateType</Reference>
    </References>
  </UAVariable>
```

(Sorry, I can't provide a full XML nodeset right now. But maybe this info is already enough.)

As preparation for `UA_Server_writeDataValue()`, my code can't find the data type by the type nodeId `AlarmStateStates` nor the type nodeId `ns=2;s=af28ed20-dbb6-5624-aa65-4b42023ab1c4` (adjusted for runtime NS index) using `UA_findDataTypeWithCustom()`. Any ideas?

paddor, 2024-03-15:
Weirdly enough, if I read the data type of an imported variable (that should have dataType=`AlarmStatesStates`) using `UA_Server_readDataType()` and `UA_findDataType()`, I sometimes get no data type or the wrong data type. It's non-deterministic.

I was hoping to use `UA_DataType.typeKind` to support these custom types but like this I have no path forward. Any ideas?

paddor, 2024-03-19:
@matkonnerth Here's a working example in C along with the full XML nodeset. It tries to get the data type of `g=16001624-6a1d-42d5-81d1-f54af84f5325`, which fails.

```server.c
#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>
#include <open62541/plugin/nodesetloader.h>

#include <signal.h>
#include <stdlib.h>
#include <stdio.h>

UA_Boolean running = true;
static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

static void
testCallback(UA_Server *server, void *data) {
  UA_StatusCode     rc;
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "testcallback");

    size_t index;
    rc = UA_Server_getNamespaceByName(server, UA_STRING("http://foobar.example.com/Instances"), &index);

    if (rc != UA_STATUSCODE_GOOD) {
      UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "could not determine namespace index for namespace URI");
      abort();
    }

    UA_Guid guid;
    UA_Guid_parse(&guid, UA_STRING("16001624-6a1d-42d5-81d1-f54af84f5325"));

    UA_NodeId nodeId = UA_NODEID_GUID(index, guid);
    UA_NodeId typeNodeId;
    rc = UA_Server_readDataType(server, nodeId, &typeNodeId);

    if (rc != UA_STATUSCODE_GOOD) {
      UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "could not read data type node id");
      abort();
    }

    UA_String typeNodeIdString;
    UA_String_init(&typeNodeIdString);
    UA_NodeId_print(&typeNodeId, &typeNodeIdString);

    UA_DataType *nodeDataType = UA_findDataType(&typeNodeId);
    /* const UA_DataTypeArray *customTypes = UA_Server_getConfig(server)->customDataTypes; */
    /* UA_DataType *nodeDataType = UA_findDataTypeWithCustom(&typeNodeId, customTypes); */
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "found data type node ID: %.*s", typeNodeIdString.length, typeNodeIdString.data);

    UA_NodeClass outNodeClass;
    rc = UA_Server_readNodeClass(server, typeNodeId, &outNodeClass);

    if (rc != UA_STATUSCODE_GOOD) {
      UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "could not read node class of data type node");
      abort();
    }

    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "data type node class: %i (is UA_NODECLASS_DATATYPE=%s)", outNodeClass, outNodeClass == UA_NODECLASS_DATATYPE ? "True" : "False");

    if (nodeDataType) {
      UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "found data type: %s, typeKind=%i", nodeDataType->typeName, nodeDataType->typeKind);
    }
    else {
      UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "could not look up data type");
      /* abort(); */
      return;
    }
[... 55 lines cut ...]
[2024-03-19 13:09:16.674 (UTC+0100)] info/server        AccessControl: Anonymous login is enabled
[2024-03-19 13:09:16.674 (UTC+0100)] warn/server        x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2024-03-19 13:09:16.674 (UTC+0100)] warn/server        x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2024-03-19 13:09:16.674 (UTC+0100)] warn/server        Maximum SecureChannels count not enough for the maximum Sessions count
[2024-03-19 13:09:16.674 (UTC+0100)] info/network       TCP     | Listening on all interfaces
[2024-03-19 13:09:16.675 (UTC+0100)] info/network       TCP 4   | Creating server socket for "0.0.0.0" on port 4840
[2024-03-19 13:09:16.677 (UTC+0100)] warn/network       TCP 4294967295  | Error opening the listen socket for "::" on port 4840 (Address family not supported by protocol)
[2024-03-19 13:09:16.764 (UTC+0100)] info/userland      Nodeset nodeset.xml loaded.
[2024-03-19 13:09:18.677 (UTC+0100)] info/userland      testcallback
[2024-03-19 13:09:18.677 (UTC+0100)] info/server        found data type node ID: ns=2;s=af28ed20-dbb6-5624-aa65-4b42023ab1c4
[2024-03-19 13:09:18.677 (UTC+0100)] info/server        data type node class: 64 (is UA_NODECLASS_DATATYPE=True)
[2024-03-19 13:09:18.677 (UTC+0100)] error/server       could not look up data type
```

Open62541 configured and installed with:
```
pwd #=> /path/to/open62541
mkdir build && cd build
cmake .. -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_NODESETLOADER=ON -DUA_ENABLE_TYPEDESCRIPTION=ON -DUA_MULTITHREADING=100 -DUA_ENABLE_PARSING=ON -DUA_ENABLE_DIAGNOSTICS=ON -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=Debug -DUA_DEBUG=1 -DCMAKE_INTERPROCEDURAL_OPTIMIZATION=OFF -DCMAKE_C_FLAGS_DEBUG='-fPIC -O0 -g'
make
sudo make install
```

The XML nodeset (gzipped):
[nodeset.xml.gz](https://github.com/open62541/open62541-nodeset-loader/files/14650379/nodeset.xml.gz)

matkonnerth, 2024-03-19:
@paddor : thank you for the detailled issue report, do you the result with the nodesetcompiler?

paddor, 2024-03-19:
@matkonnerth Sorry, I didn't understand the question. The nodeset itself works, in another UA stack and also in this UA stack, at least when checked via UaExpert. What doesn't work is getting the real data type of the variable node.

matkonnerth, 2024-03-19:
sry, the open62541 has also the ability to generate c code for the information model (done with the nodesetcompiler). Would be interesting if the issue preserves with that method.

paddor, 2024-03-19:
Here's the output:

```
$ python3 ./nodeset_compiler.py --types-array=UA_TYPES --existing ../../deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml --xml /home/roadster/dev/roadster-projects/internal/roadster-protocol-simulator/conf/nodeset.xml myNS
INFO:__main__:Preprocessing (existing) ../../deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing /home/roadster/dev/roadster-projects/internal/roadster-protocol-simulator/conf/nodeset.xml
INFO:__main__:Generating Code for Backend: open62541
Traceback (most recent call last):
  File "/home/roadster/dev/oss/open62541-ruby/ext/open62541/deps/open62541/tools/nodeset_compiler/./nodeset_compiler.py", line 227, in <module>
    generateOpen62541Code(ns, args.outputFile, args.internal_headers, args.typesArray)
  File "/home/roadster/dev/oss/open62541-ruby/ext/open62541/deps/open62541/tools/nodeset_compiler/backend_open62541.py", line 231, in generateOpen62541Code
    code = generateNodeCode_begin(node, nodeset, code_global)
  File "/home/roadster/dev/oss/open62541-ruby/ext/open62541/deps/open62541/tools/nodeset_compiler/backend_open62541_nodes.py", line 532, in generateNodeCode_begin
    code.append(generateNodeIdCode(node.id) + ",")
  File "/home/roadster/dev/oss/open62541-ruby/ext/open62541/deps/open62541/tools/nodeset_compiler/backend_open62541_datatypes.py", line 102, in generateNodeIdCode
    raise Exception(str(value) + " no NodeID generation for bytestring and guid..")
Exception: ns=2;g=fc658c51-781b-4c13-8d4f-8c45f4054389 no NodeID generation for bytestring and guid..
```

What does this mean?

matkonnerth, 2024-03-19:
oh, seems that NodeIds with type ByteString or Guid are not supported by the nodesetcompiler

paddor, 2024-03-19:
I've added support for GUID node IDs in a PR: https://github.com/open62541/open62541/pull/6355

The script completes successfully and gives me two files.

paddor, 2024-03-19:
Generated header file:
```myNS.h
/* WARNING: This is a generated file.
 * Any manual changes will be overwritten. */

#ifndef MYNS_H_
#define MYNS_H_

#ifdef UA_ENABLE_AMALGAMATION
# include "open62541.h"
#else
# include <open62541/server.h>
#endif

_UA_BEGIN_DECLS

extern UA_StatusCode myNS(UA_Server *server);

_UA_END_DECLS

#endif /* MYNS_H_ */
```

Generated C file is attached. Using the two new files in this program:

```serverWithmyNS.c
#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>
#include <open62541/plugin/nodesetloader.h>

#include <signal.h>
#include <stdlib.h>
#include <stdio.h>

#include "myNS.h"

UA_Boolean running = true;
static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

static void
testCallback(UA_Server *server, void *data) {
  UA_StatusCode     rc;
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "testcallback");

    size_t index;
    rc = UA_Server_getNamespaceByName(server, UA_STRING("http://foobar.example.com/Instances"), &index);

    if (rc != UA_STATUSCODE_GOOD) {
      UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "could not determine namespace index for namespace URI");
      abort();
    }

    UA_Guid guid;
    UA_Guid_init(&guid);
    UA_Guid_parse(&guid, UA_STRING("16001624-6a1d-42d5-81d1-f54af84f5325"));

    UA_NodeId nodeId = UA_NODEID_GUID(index, guid);
    UA_NodeId typeNodeId;
    UA_NodeId_init(&typeNodeId);
    rc = UA_Server_readDataType(server, nodeId, &typeNodeId);

    if (rc != UA_STATUSCODE_GOOD) {
      UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "could not read data type node id");
      abort();
    }

[... 77 lines cut ...]

Output:
```
$ ./serverWithmyNS
[2024-03-19 17:16:23.880 (UTC+0100)] info/eventloop     Starting the EventLoop
[2024-03-19 17:16:23.880 (UTC+0100)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2024-03-19 17:16:23.880 (UTC+0100)] info/server        AccessControl: Anonymous login is enabled
[2024-03-19 17:16:23.880 (UTC+0100)] warn/server        x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2024-03-19 17:16:23.981 (UTC+0100)] warn/userland      ServerUrls already set. Overriding.
[2024-03-19 17:16:23.981 (UTC+0100)] warn/server        AccessControl: Unconfigured Acces
[myNS.c.gz](https://github.com/open62541/open62541-nodeset-loader/files/14653411/myNS.c.gz)
sControl. Users have all permissions.
[2024-03-19 17:16:23.981 (UTC+0100)] info/server        AccessControl: Anonymous login is enabled
[2024-03-19 17:16:23.981 (UTC+0100)] warn/server        x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2024-03-19 17:16:23.981 (UTC+0100)] warn/server        x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2024-03-19 17:16:23.981 (UTC+0100)] warn/server        Maximum SecureChannels count not enough for the maximum Sessions count
[2024-03-19 17:16:23.981 (UTC+0100)] info/network       TCP     | Listening on all interfaces
[2024-03-19 17:16:23.983 (UTC+0100)] info/network       TCP 4   | Creating server socket for "0.0.0.0" on port 4840
[2024-03-19 17:16:23.985 (UTC+0100)] warn/network       TCP 4294967295  | Error opening the listen socket for "::" on port 4840 (Address family not supported by protocol)
[2024-03-19 17:16:23.987 (UTC+0100)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode: Node could not add the new node to the nodestore with error code BadNodeIdExists
[2024-03-19 17:16:23.987 (UTC+0100)] error/server       Could not add the example nodeset. Check previous output for any error.
fish: Job 2, './serverWithmyNS' terminated by signal SIGABRT (Abort)
```

It seems the compiled nodeset tries to create the same node twice?

paddor, 2024-03-19:
@matkonnerth I've added logging to the generated code so I can see which node ID causes trouble. 

```
[2024-03-19 18:10:38.829 (UTC+0100)] info/eventloop	Starting the EventLoop
[2024-03-19 18:10:38.829 (UTC+0100)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
[2024-03-19 18:10:38.829 (UTC+0100)] info/server	AccessControl: Anonymous login is enabled
[2024-03-19 18:10:38.829 (UTC+0100)] warn/server	x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2024-03-19 18:10:38.927 (UTC+0100)] warn/userland	ServerUrls already set. Overriding.
[2024-03-19 18:10:38.927 (UTC+0100)] warn/server	AccessControl: Unconfigured AccessControl. Users have all permissions.
[2024-03-19 18:10:38.927 (UTC+0100)] info/server	AccessControl: Anonymous login is enabled
[2024-03-19 18:10:38.927 (UTC+0100)] warn/server	x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2024-03-19 18:10:38.927 (UTC+0100)] warn/server	x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2024-03-19 18:10:38.927 (UTC+0100)] warn/server	Maximum SecureChannels count not enough for the maximum Sessions count
[2024-03-19 18:10:38.927 (UTC+0100)] trace/session	TCP 0	| SC 0	| Session "Administrator"	| Write attribute 13 of Node i=2254
[2024-03-19 18:10:38.927 (UTC+0100)] trace/session	TCP 0	| SC 0	| Session "Administrator"	| Write attribute 13 of Node i=2257
[2024-03-19 18:10:38.927 (UTC+0100)] info/network	TCP	| Listening on all interfaces
[2024-03-19 18:10:38.929 (UTC+0100)] info/network	TCP 4	| Creating server socket for "0.0.0.0" on port 4840
[2024-03-19 18:10:38.931 (UTC+0100)] warn/network	TCP 4294967295	| Error opening the listen socket for "::" on port 4840 (Address family not supported by protocol)
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=f9df004f-da1a-5d1c-976e-ea4135ffc8a7
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=f9afc88c-c431-5dfc-9122-060476e433d8
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=2226ed75-a294-5888-820f-d04ca325cdfc
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=f7a88a1a-2a3e-5148-94bb-2bd6bdd5f0e0
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=68734b65-41e6-5bfc-89d8-2d07b0a25129
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=e2535265-b87b-5a9b-8ab1-3f639dd44f60
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=d7e89415-4944-5c3c-b9fb-5390ad9728fa
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=444bf352-8ff2-501e-a684-72768b961da5
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=e01120e0-f442-5524-8aa2-b84940e75f92
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=2a9764a5-2f14-5453-9c26-95e939c43957
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=d6c9f0f0-b61a-56c7-a8a0-48093bbbbe3b
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=fe0ea47c-eb37-5bf1-9427-df9a853374ee
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=ad28882d-21cb-503e-a960-d278b8dfe822
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=f7b7bab9-7554-5620-b20c-b6d71f3ad378
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=2d910abe-b73e-5f29-a8e5-50f8d34e7e16
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=ba574a1b-5202-504f-8de3-46b6b5101170
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=44215d48-b38c-580a-9791-42c07f5b596d
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=7f4a9352-0ca8-5768-86d1-e628f7a83c9b
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=d8e2a7e5-f76c-5bc1-a9e4-af7f0d48ff93
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=cc8d9c94-4804-5997-9ee2-dd57c6d5d4ff
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=baed81e7-bc19-5a22-97c5-78f03d166c49
[2024-03-19 18:10:38.931 (UTC+0100)] info/userland	adding node: ns=1;s=f7d25876-5dc8-5349-ac0a-469bd8d62528
[2024-03-19 18:10:38.932 (UTC+0100)] info/userland	adding node: ns=1;s=01090c8f-ea39-5fb4-816c-3561fa2f4a51
[2024-03-19 18:10:38.932 (UTC+0100)] info/userland	adding node: ns=1;s=59cf69e1-ee56-5e8c-829c-5234c9719536
[2024-03-19 18:10:38.932 (UTC+0100)] info/userland	adding node: ns=1;s=7845c676-a0fe-57f4-ab82-7bc4bcbb86da
[2024-03-19 18:10:38.932 (UTC+0100)] info/userland	adding node: ns=1;s=13fe661c-9ace-5b82-b506-44308f5c1b10
[2024-03-19 18:10:38.932 (UTC+0100)] info/userland	adding node: ns=1;s=84286e72-4b8d-50e3-bd44-c38c68ffc71d
[2024-03-19 18:10:38.932 (UTC+0100)] info/userland	adding node: ns=1;s=17842011-afc6-50ad-bded-0ec28f211116
[2024-03-19 18:10:38.932 (UTC+0100)] info/userland	adding node: ns=1;s=131cddbf-7692-5049-a2d9-a3253028330e
[2024-03-19 18:10:38.932 (UTC+0100)] info/userland	adding node: ns=1;s=757dc22c-bc04-5746-8344-1bc90b4e7126
[2024-03-19 18:10:38.932 (UTC+0100)] info/userland	adding node: ns=1;s=31ba064f-12b8-56f4-8ce2-0704e9edc9b7
[2024-03-19 18:10:38.932 (UTC+0100)] info/userland	adding node: ns=1;s=93331a32-82e7-5662-a17a-f1bcbc52297d
[2024-03-19 18:10:38.932 (UTC+0100)] info/userland	adding node: ns=1;s=8cda075b-df3a-5d63-a1d1-0d1de06a709e
[... cut ...]
[2024-03-19 18:10:38.933 (UTC+0100)] info/session	TCP 0	| SC 0	| Session "Administrator"	| AddNode: Node could not add the new node to the nodestore with error code BadNodeIdExists
[2024-03-19 18:10:38.933 (UTC+0100)] error/server	Could not add the example nodeset. Check previous output for any error.
fish: Process 270262, './serverWithmyNS' from job 2, './serverWithmyNS | sed -r "s/\x…' terminated by signal SIGABRT (Abort)
```

It is always the same: `ns=2;g=c99a4933-9ba-456e-89c4-b0ee01003c89` 
I'm not sure if it has to do with the missing leading zeroes in the GUID.

paddor, 2024-03-22:
I've fixed the issue with missing leading zeroes. See https://github.com/open62541/open62541/pull/6355.

The server with the compiled nodeset now imports many more variables before aborting:

```
...
[2024-03-22 13:07:22.748 (UTC+0100)] info/userland      adding node: ns=1;s=0227d833-2a48-4578-9c95-ac4600f8314c
[2024-03-22 13:07:22.748 (UTC+0100)] info/userland      adding node: ns=1;s=0168852c-858e-4407-8dbd-ac4600f8313e
[2024-03-22 13:07:22.748 (UTC+0100)] info/userland      adding node: ns=1;s=015ef1e0-d393-4631-8a52-ac4600f8313e
[2024-03-22 13:07:22.748 (UTC+0100)] info/userland      adding node: ns=1;s=0152d396-e7c5-4086-908e-ac4600f83139
[2024-03-22 13:07:22.748 (UTC+0100)] info/userland      adding node: ns=1;s=002642a4-dc5d-432a-b8ba-ac69012aea6c
[2024-03-22 13:07:22.748 (UTC+0100)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Read attribute 13 of Node ns=2;s=002642a4-dc5d-432a-b8ba-ac69012aea6c
[2024-03-22 13:07:22.748 (UTC+0100)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=2;s=002642a4-dc5d-432a-b8ba-ac69012aea6c): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[2024-03-22 13:07:22.748 (UTC+0100)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Write attribute 13 of Node ns=2;s=002642a4-dc5d-432a-b8ba-ac69012aea6c
[2024-03-22 13:07:22.748 (UTC+0100)] trace/session      TCP 0   | SC 0  | Session "Administrator"       | Read attribute 8 of Node i=29
[2024-03-22 13:07:22.748 (UTC+0100)] warn/server        Writing the value of Node ns=2;s=002642a4-dc5d-432a-b8ba-ac69012aea6c failed with the following reason: DataType of the value is incompatible
[2024-03-22 13:07:22.748 (UTC+0100)] info/session       TCP 0   | SC 0  | Session "Administrator"       | WriteRequest returned status code BadTypeMismatch
[2024-03-22 13:07:22.748 (UTC+0100)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=2;s=002642a4-dc5d-432a-b8ba-ac69012aea6c): Could not create a default value with StatusCode BadTypeMismatch
[2024-03-22 13:07:22.748 (UTC+0100)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=2;s=002642a4-dc5d-432a-b8ba-ac69012aea6c): Type-checking failed with error code BadTypeMismatch
[2024-03-22 13:07:22.748 (UTC+0100)] error/server       Could not add the example nodeset. Check previous output for any error.
fish: Job 3, './serverWithmyNS' terminated by signal SIGABRT (Abort)
```

If that error message is correct, a default value is missing (originally in the XML nodeset, but here in the compiled nodeset). That would mean the nodesetLoader should complain about the missing default value too, instead of silently importing the nodeset without warnings/errors. Correct?

paddor, 2024-03-22:
To reproduce, use [issue_253.tgz](https://github.com/open62541/open62541-nodeset-loader/files/14725715/issue_253.tgz) with the following content:
```
issue253/
issue253/serverWithmyNS.c
issue253/myNS.h
issue253/serverWithmyNS
issue253/myNS.c
issue253/server
issue253/nodeset.xml
issue253/server.c
issue253/Makefile
```

jpfr, 2026-09-05:
Imported runtime DataTypes are now owned and registered by the server and can be resolved with UA_Server_findDataType. The server-side fix that also uses these internal custom types during attribute writes is being backported to open62541 1.5 in open62541/open62541#8383. Closing as fixed.

closed 2026-09-05 by jpfr