open62541/open62541-nodeset-loader #255: About static hierarchical reference...

xydan83, 2024-03-26:
Hi!

I tried to find how you sort links by hierarchy or not and found some repository...

https://github.com/open62541/open62541-nodeset-loader/blob/c947a55d976b5d47fa6e52d00b592b279b371a82/src/InternalRefService.c#L23

Is it a statistical hierarchical storage for sorting?

matkonnerth, 2024-03-29:
this is an array of the known hierachical references from namespace 0. In normal operation the "ReferenceService" uses the references from the information model: https://github.com/open62541/open62541-nodeset-loader/blob/9967a457a919d4219185d1582b0b99661471bb13/backends/open62541/src/RefServiceImpl.c#L162

xydan83, 2024-03-29:
I understand. Thank you!

closed 2024-03-29 by xydan83