open62541/open62541-nodeset-loader #256: Error building nodesetloader in Open62541 library version 1.4.0.

xydan83, 2024-04-12:
Hi!
I'll duplicate the problem with Open62541 v1.4.0 here just in case.
https://github.com/open62541/open62541/issues/6401

matkonnerth, 2024-04-13:
I've created a PR to fix this issue, maybe you can test the fix before merging.

xydan83, 2024-04-15:
> I've created a PR to fix this issue, maybe you can test the fix before merging.

I checked your PR, tried to build the project (cmake) and compile - it works) Thank you! I hope that this fix will appear in the new version (1.4.1) of the Open62541 library.

xydan83, 2024-04-15:
There is something wrong with the **GccDebugMemcheck** test.
[image]

[image]

https://github.com/open62541/open62541-nodeset-loader/actions/runs/8686888886/job/23819279363