open62541/open62541-nodeset-loader #266: [BUG] Opc.Ua.Sample/TestData not working

electro-logic, 2024-11-14:
Please see https://github.com/open62541/open62541/issues/6867

matkonnerth, 2024-11-18:
please test with this PR https://github.com/open62541/open62541-nodeset-loader/pull/267

electro-logic, 2024-11-19:
Hi,

I think would be better to add an error message or something to warn users about unsupported schema.

I tried NodeSet2.xml but I get the following error:

```
[2024-11-19 14:28:24.949 (UTC+0100)] debug/userland     Start import nodeset: TestData.NodeSet2.xml
[2024-11-19 14:28:24.960 (UTC+0100)] error/userland     node with unresolved reference(s): NodeId(ns=2;i=10175)
[2024-11-19 14:28:24.961 (UTC+0100)] error/userland     importing the nodeset failed, nodes were not added
Nodeset TestData.NodeSet2.xml could not be loaded, exit
```

Thanks

matkonnerth, 2024-11-19:
have you built the open62541 with full namespace 0?

electro-logic, 2024-11-19:
I think was compiled with the reduced namespace. Is nodeset-loader working with the full configuration only?

matkonnerth, 2024-11-19:
seems that the TestData.NodeSet2.xml has references to nodes from namespace 0, which are not compiled in with the reduced configuration.

electro-logic, 2024-11-19:
In this case would be good to log the problematic node without aborting the whole import operation. Or at least have this option.

Another improvement  can be about the logging, because the LogLevel 0 is not part of the UA_LogLevel enum, I have to use this code:

```
    UA_Server* server = UA_Server_new();
    auto config = UA_Server_getConfig(server);
    config->allowNonePolicyPassword = true;
    config->logging = UA_Log_Stdout_new((UA_LogLevel)0);
    UA_ServerConfig_setDefault(config);
```

But I think that having everything logged at LogLevel 100 should be okay and easier to use.

closed 2024-11-20 by matkonnerth

electro-logic, 2024-11-20:
> seems that the TestData.NodeSet2.xml has references to nodes from namespace 0, which are not compiled in with the reduced configuration.

I'm gonna test with the full namespace 0 today

electro-logic, 2024-11-20:
I can't compile with the full namespace, i get this exception from the branch 1.4:

Exception: Reference ns=1;i=15001<--[ns=0;i=47]--ns=0;i=11715 has an unknown target

note: reduced namespace is compiling fine

```
[1/1] Cleaning all built files...
  Cleaning... 104 files.
  
  [1/194] Generating src_generated/open62541/statuscodes.h, src_generated/open62541/statuscodes.c
  [2/194] Generating ../../src_generated/open62541/di_nodeids.h
  [3/194] Generating ../../src_generated/open62541/types_di_generated.c, ../../src_generated/open62541/types_di_generated.h, ../../src_generated/open62541/types_di_generated_handling.h
  [4/194] Generating src_generated/open62541/nodeids.h
  [5/194] Generating src_generated/open62541/transport_generated.c, src_generated/open62541/transport_generated.h, src_generated/open62541/transport_generated_handling.h
  [6/194] Generating ../../src_generated/open62541/types_testnodeset_generated.c, ../../src_generated/open62541/types_testnodeset_generated.h, ../../src_generated/open62541/types_testnodeset_generated_handling.h
  [7/194] Generating src_generated/open62541/types_generated.c, src_generated/open62541/types_generated.h, src_generated/open62541/types_generated_handling.h
  [8/194] Generating ../../src_generated/open62541/testnodeset_nodeids.h
  [9/194] Generating ../../src_generated/open62541/powerlink_nodeids.h
  [10/194] Generating ../../src_generated/open62541/example_nodeids.h
  [11/194] Generating src_generated/open62541/namespace0_generated.c, src_generated/open62541/namespace0_generated.h
  INFO:__main__:Preprocessing C:/Projects/open62541/tools/schema/Opc.Ua.NodeSet2.Reduced.xml
  
  INFO:__main__:Generating Code for Backend: open62541
  
  INFO:__main__:NodeSet generation code successfully printed
  
  [12/194] Generating ../../src_generated/open62541/namespace_example_generated.c, ../../src_generated/open62541/namespace_example_generated.h
  INFO:__main__:Preprocessing (existing) C:/Projects/open62541/examples/nodeset/../../tools/schema/Opc.Ua.NodeSet2.Reduced.xml
  
  INFO:__main__:Preprocessing C:/Projects/open62541/examples/nodeset/server_nodeset.xml
  
  INFO:__main__:Generating Code for Backend: open62541
  
  INFO:__main__:NodeSet generation code successfully printed
  
  [13/194] Generating ../../src_generated/open62541/namespace_di_generated.c, ../../src_generated/open62541/namespace_di_generated.h
  FAILED: src_generated/open62541/namespace_di_generated.c src_generated/open62541/namespace_di_generated.h C:/Projects/open62541/out/build/x64-Debug/src_generated/open62541/namespace_di_generated.c C:/Projects/open62541/out/build/x64-Debug/src_generated/open62541/namespace_di_generated.h 
  C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Projects\open62541\out\build\x64-Debug\examples\nodeset && "C:\Program Files\Python313\python.exe" C:/Projects/open62541/tools/nodeset_compiler/nodeset_compiler.py --internal-headers --bsd=C:/Projects/open62541/deps/ua-nodeset/DI/Opc.Ua.Di.Types.bsd --types-array=UA_TYPES --types-array=UA_TYPES_DI --existing=C:/Projects/open62541/tools/schema/Opc.Ua.NodeSet2.Reduced.xml --xml=C:/Projects/open62541/deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml C:/Projects/open62541/out/build/x64-Debug/src_generated/open62541/namespace_di_generated"
  INFO:__main__:Preprocessing (existing) C:/Projects/open62541/tools/schema/Opc.Ua.NodeSet2.Reduced.xml
  INFO:__main__:Preprocessing C:/Projects/open62541/deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml
  ['http://opcfoundation.org/UA/', 'http://opcfoundation.org/UA/DI/']
  Traceback (most recent call last):
    File "C:\Projects\open62541\tools\nodeset_compiler\nodeset_compiler.py", line 187, in <module>
      ns.sanitize()
      ~~~~~~~~~~~^^
    File "C:\Projects\open62541\tools\nodeset_compiler\nodeset.py", line 137, in sanitize
      raise Exception("Reference " + str(ref) + " has an unknown target")
  Exception: Reference ns=1;i=15001<--[ns=0;i=47]--ns=0;i=11715 has an unknown target
  
  [14/194] Generating ../../src_generated/open62541/namespace_testnodeset_generated.c, ../../src_generated/open62541/namespace_testnodeset_generated.h
  FAILED: src_generated/open62541/namespace_testnodeset_generated.c src_generated/open62541/namespace_testnodeset_generated.h C:/Projects/open62541/out/build/x64-Debug/src_generated/open62541/namespace_testnodeset_generated.c C:/Projects/open62541/out/build/x64-Debug/src_generated/open62541/namespace_testnodeset_generated.h 
  C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Projects\open62541\out\build\x64-Debug\examples\nodeset && "C:\Program Files\Python313\python.exe" C:/Projects/open62541/tools/nodeset_compiler/nodeset_compiler.py --internal-headers --bsd=C:/Projects/open62541/examples/nodeset/testtypes.bsd --types-array=UA_TYPES --types-array=UA_TYPES_TESTNODESET --existing=C:/Projects/open62541/tools/schema/Opc.Ua.NodeSet2.Reduced.xml --xml=C:/Projects/open62541/examples/nodeset/testnodeset.xml C:/Projects/open62541/out/build/x64-Debug/src_generated/open62541/namespace_testnodeset_generated"
  INFO:__main__:Preprocessing (existing) C:/Projects/open62541/tools/schema/Opc.Ua.NodeSet2.Reduced.xml
  INFO:__main__:Preprocessing C:/Projects/open62541/examples/nodeset/testnodeset.xml
  ['http://opcfoundation.org/UA/', 'http://yourorganisation.org/test/']
  Traceback (most recent call last):
    File "C:\Projects\open62541\tools\nodeset_compiler\nodeset_compiler.py", line 187, in <module>
      ns.sanitize()
      ~~~~~~~~~~~^^
[... cut ...]
  [15/194] Building C object CMakeFiles\open62541-object.dir\src\ua_types_encoding_binary.c.obj
  [16/194] Building C object CMakeFiles\open62541-object.dir\src\ua_types.c.obj
  [17/194] Building C object CMakeFiles\open62541-object.dir\src_generated\open62541\types_generated.c.obj
  [18/194] Generating ../../src_generated/open62541/types_powerlink_generated.c, ../../src_generated/open62541/types_powerlink_generated.h, ../../src_generated/open62541/types_powerlink_generated_handling.h
  ninja: build stopped: subcommand failed.
```

electro-logic, 2024-11-24:
Now that the FULL ns0 is compiling I tried again the PR. 

TestData.NodeSet2.xml is now imported, with some errors:

```
[2024-11-25 00:08:47.365 (UTC+0100)] debug/userland     Start import nodeset: TestData.NodeSet2.xml
[2024-11-25 00:08:47.374 (UTC+0100)] debug/userland     imported ReferenceTypes: 0
[2024-11-25 00:08:47.374 (UTC+0100)] debug/userland     imported DataTypes: 29
[2024-11-25 00:08:47.374 (UTC+0100)] debug/userland     imported ObjectTypes: 10
[2024-11-25 00:08:47.374 (UTC+0100)] debug/userland     imported VariableTypes: 0
[2024-11-25 00:08:47.374 (UTC+0100)] debug/userland     imported Objects: 52
[2024-11-25 00:08:47.374 (UTC+0100)] debug/userland     imported Methods: 88
[2024-11-25 00:08:47.385 (UTC+0100)] debug/userland     imported Variables: 901
[2024-11-25 00:08:47.385 (UTC+0100)] debug/userland     imported Views: 0
[2024-11-25 00:08:47.385 (UTC+0100)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | Cannot add reference - target ns=2;i=10372 does not exist
[2024-11-25 00:08:47.385 (UTC+0100)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | Cannot add reference - target ns=2;i=10451 does not exist
[2024-11-25 00:08:47.385 (UTC+0100)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | Cannot add reference - target ns=2;i=10530 does not exist
[2024-11-25 00:08:47.385 (UTC+0100)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | Cannot add reference - target ns=2;i=10665 does not exist
[2024-11-25 00:08:47.385 (UTC+0100)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | Cannot add reference - target ns=2;i=10204 does not exist
[2024-11-25 00:08:47.385 (UTC+0100)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | Cannot add reference - target ns=2;i=10288 does not exist
[2024-11-25 00:08:47.385 (UTC+0100)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | Cannot add reference - target ns=2;i=11000 does not exist
[2024-11-25 00:08:47.385 (UTC+0100)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | Cannot add reference - target ns=2;i=11079 does not exist
[2024-11-25 00:08:47.385 (UTC+0100)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | Cannot add reference - target ns=2;i=11158 does not exist
[2024-11-25 00:08:47.385 (UTC+0100)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | Cannot add reference - target ns=2;i=11293 does not exist
[2024-11-25 00:08:47.385 (UTC+0100)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | Cannot add reference - target ns=2;i=10832 does not exist
[2024-11-25 00:08:47.385 (UTC+0100)] debug/session      TCP 0   | SC 0  | Session "Administrator"       | Cannot add reference - target ns=2;i=10916 does not exist
```

The strange thing for these nodes is the namespace, the xml file is mentioning ns=1 while the warning is mentioning ns=2.

Example:

<Reference ReferenceType="HasTrueSubState">**ns=1**;i=10372</Reference>

Cannot add reference - target **ns=2**;i=10372 does not exist