open62541/open62541-nodeset-loader #273: Cloning of open62541 fails when trying to access cmake-conan via ssh

ernstblechaPT, 2024-12-19:
In my build process I try to clone the open62541-repository with all the submodules resolved on a Gitlab-Runner. Until this change

https://github.com/open62541/open62541-nodeset-loader/blob/fcccfc71c97c2292bd6fcff84329cd505668310e/.gitmodules#L3

this worked.

After this change my clone fails with the following error:

```
Cloning into '/builds/test/open62541/deps/nodesetLoader/cmake-conan'...
error: cannot run ssh: No such file or directory
fatal: unable to fork
fatal: clone of 'git@github.com:conan-io/cmake-conan.git' into submodule path '/builds/test/open62541/deps/nodesetLoader/cmake-conan' failed
Failed to clone 'cmake-conan'. Retry scheduled
Cloning into '/builds/test/open62541/deps/nodesetLoader/cmake-conan'...
error: cannot run ssh: No such file or directory
fatal: unable to fork
fatal: clone of 'git@github.com:conan-io/cmake-conan.git' into submodule path '/builds/test/open62541/deps/nodesetLoader/cmake-conan' failed
Failed to clone 'cmake-conan' a second time, aborting
fatal: Failed to recurse into submodule path 'open62541/deps/nodesetLoader'
```

I am aware that the better solution would be to get the gitlab runner docker image fixed - but still wanted to ask if there is a specific reason to have this dependency listed with the ssh-url or if this could be changed to a https url (as all other submodules are listed with the https-url)?

Thank you for your answer

matkonnerth, 2024-12-19:
should be no problem to switch to https: https://github.com/open62541/open62541-nodeset-loader/pull/274

matkonnerth, 2024-12-19:
merged it to master, please try again

ernstblechaPT, 2024-12-19:
Thank you for the quick solution! Works again

closed 2024-12-19 by ernstblechaPT

matkonnerth, 2024-12-19:
thanks for testing, don't hesitate to propose PRs if something isn't working as expected.