open62541/open62541-nodeset-loader #285: Deprecated parser!

OgreTransporter, 2025-05-02:
https://github.com/open62541/open62541-nodeset-loader/blob/fbedde3f81c6c517ce0656661c13e46503b19342/src/Parser.c#L10

Please, have a look at the source file: https://github.com/GNOME/libxml2/blob/master/include/libxml/SAX.h
```
/*
 * Summary: Old SAX version 1 handler, deprecated
 * Description: DEPRECATED set of SAX version 1 interfaces used to
 *              build the DOM tree.
 *
 * Copy: See Copyright for the status of this software.
 *
 * Author: Daniel Veillard
 */

#ifndef __XML_SAX_H__
#define __XML_SAX_H__

#ifdef __GNUC__
  #warning "libxml/SAX.h is deprecated"
#endif

#endif /* __XML_SAX_H__ */
```

The functions have been removed on Jun 17, 2024!

jpfr, 2026-09-05:
The loader was integrated into open62541 by open62541/open62541#8381 and switched from libxml2 SAX to the vendored yxml parser. The deprecated parser path no longer exists. Closing as fixed.

jpfr, 2026-09-05:
The required yxml and XML-decoder changes are also being backported to open62541 1.5 in open62541/open62541#8383.

closed 2026-09-05 by jpfr